stereolabs/zed-ros-wrapper #524: Object detection error ZED 2 corrupted sdk installation

elisastefanini, 2020-02-17:
Hi!
I have a ZED 2 camera working with ros kinetic on ubuntu 16.04. I installed sdk and cuda following this link: https://www.stereolabs.com/docs/installation/linux/ 
I would like to use the obstacle detection module, so I set object_detection/od_enabled in params/zed2.yaml  to true but when I start the launch file roslaunch zed_wrapper zed2.launch
I get this error:
[ INFO] [1581953615.691354308]: *** Starting Object Detection ***
[ERROR] [1581953615.694651692]: Object detection error: CORRUPTED SDK INSTALLATION

How can I fix this problem? 
Best regards
Elisa

Myzhar, 2020-02-17:
Hi @elisastefanini 
is your PC connected to internet? The Object Detection module must download a few data from our server the first time it starts.
Please also verify that the folder `/usr/local/zed/resources/` contains a model, if not try to reinstall the SDK again and reboot the system when the installation is completed.

Before proceeding with the above tests can you please run the diagnostic tool and send it to us to support@stereolabs.com?

elisastefanini, 2020-02-18:
Hi @Myzhar 
yes my laptop is connected to internet.
I've check the folder and it contains object_1.2.model.
I've sent the diagnostic tool result to support@stereolabs.com
Should I try to reinstall SDK?

Myzhar, 2020-02-18:
Yes, you can try to reinstall it. Thank you

elisastefanini, 2020-02-18:
Yes, now it's working! Thank you

Myzhar, 2020-02-18:
You are welcome :+1:

closed 2020-02-18 by Myzhar

HystericalHilb, 2021-05-02:
@Myzhar I use SDK3.5 of ZED2 with cuda11.1, I have tried all of what you have said above, but I can't solve this porblem.
-------------------------------------------------------
**here's cuda11.1**
PS C:\Users\34960> nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Tue_Sep_15_19:12:04_Pacific_Daylight_Time_2020
Cuda compilation tools, release 11.1, V11.1.74
Build cuda_11.1.relgpu_drvr455TC455_06.29069683_0
-------------------------------------------------------------------
**here's zed diagnostic's result, there may be something wrong resulting in SDK error**
"ZED SDK": {
        "CUDA_reachable": false,
        "driverVersion": "457.09",
        "drivervDllVersion": "27.21.14.5709",
        "error": [
            "<b>CUDA 11.1 not ready</b> <br/> Your computer needs to be restarted."
        ],
        "loaded": false,
        "runtimeDllVersion": 11010
    }
----------------------------------------------------------------
what can I do for this, hope you can help me. thanks

Myzhar, 2021-05-02:
Hi @HystericalHilb 
Please do not cross-post and ask for help in closed issues.
Have you tried to uninstall and re-install the Nvidia Driver, CUDA and the ZED SDK?

HystericalHilb, 2021-05-02:
I have tried

anyuser, 2021-06-18:
same problem here. I uninstalled/reinstalled both CUDA and ZED SDK multiple times, but I'm still getting the same error. I previously had ZED SDK 3.4 installed (which worked), in case that matters.

Myzhar, 2021-06-19:
Try to delete the folder `/usr/local/zed` and then reinstall the SDK. It will ask to download again the AI models the first time you use them

anyuser, 2021-06-19:
Actually I had the problem on Windows, the Linux installation worked for me without issues. I was able to fix it on Windows by uninstalling/reinstalling a few more times, while rebooting the computer in between. Not sure which step exactly solved it, but I suspect CUDA itself was the problem, not the ZED sdk.

Myzhar, 2021-06-19:
Are you using the ROS wrapper under Windows?

anyuser, 2021-06-22:
> Are you using the ROS wrapper under Windows?

No, probably not, just tried to get the Zed SDK to work and this was the only issue I found with a similar problem description.

mgzjzzhang, 2021-12-18:
I delete the folder /usr/local/zed and reinstall the SDK many times,but still error:SL_AI_NOT_FOUND  
Object detection error: CORRUPTED SDK INSTALLATION,and the the folder /usr/local/zed/resources/ is empty