stereolabs/zed-ros-wrapper #525: Zedcam update rate in ROS is slow(.1Hz), but fast when using the SDK examples

UltharPeliki7, 2020-02-19:
As the title says, the update rate in ROS is abysmally slow. It used to be much faster, but after updating to the 3.0 SDK and also updating to Nvidia 430 drivers, I no longer have fast updates in ROS. This issue is present whether viewing both pointcloud data as well as camera output in RVIZ. I have tried changing resolution/framerate/quality of the camera through the yaml files, but I suspect the problem is elsewhere. Any help you can provide would be greatly appreciated.

System is ROS Kinetic on Ubuntu 16.04 with NVIDIA-Driver Version: 430.64   CUDA Version: 10.
Graphics card is a GTX 750ti, while monitoring system resources my processor remains under 10% utilization, 4gb/16gb ram, and typically 1-1.5gb/2gb VRAM.

Myzhar, 2020-02-19:
Hi @UltharPeliki7 
I need more info to understand the origin of you problem:
- ROS version
- Local or remote subscribing?
- How do you measure framerate?

UltharPeliki7, 2020-02-19:
ROS version is the Kinetic LTS, I should be local subscribing, I'm not sure what remote subscribing is(I am not doing anything intersystem), I am measuring framerate by seeing how often it updates both through a rostopic echo, and through looking at the RVIZ viewer. I have it set to 30fps, and am receiving updates on all other systems at expected speeds(lidar, ultrasound, GPS all are fine), but the zedcam isn't. I have tried turning off all other components(by both physical disconnection and by making sure there were no topics.

Myzhar, 2020-02-19:
Can you publish the result of the command
`rostopic hz <topic_name>`
for an image topic and for the pointcloud?

Further questions:
- Are you using the latest version of the ROS Wrapper?
- Which camera model are you using?

UltharPeliki7, 2020-02-19:
`rostopic hz /zed/zed_node/point_cloud/cloud_registered
subscribed to [/zed/zed_node/point_cloud/cloud_registered]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 0.125
min: 8.028s max: 8.028s std dev: 0.00000s window: 2`

rostopic hz /zed/zed_node/left/image_rect_color
subscribed to [/zed/zed_node/left/image_rect_color]
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
no new messages
average rate: 0.124
	min: 8.039s max: 8.039s std dev: 0.00000s window: 2

Yes, I downloaded the wrapper last week. I am using the first model zedcam known as Zed

Myzhar, 2020-02-19:
Ok, how are you starting the node?
1) `$ roslaunch zed_wrapper zed.launch`
2) `$ roslaunch zed_display_rviz display_zed.launch`

UltharPeliki7, 2020-02-19:
I typically launch it using 1, but I just tried number 2 and resulted in the same problem.

Myzhar, 2020-02-19:
Ok, I just tried using a ZED to be sure that there was not a regression in the code with the latest changes and everything works correctly with my setup.
I'm on Ubuntu 18 using ROS Melodic, but there are not as many differences to justify such a problem.

I ask you to run `ZED_Explorer` and `ZED_Depth_Viewer` to verify that everything works fine with them. In positive case I will investigate further about any possible problem with Ubuntu 16 and ROS Kinetic.

Another test that you can perform is to delete the `devel` and `build` folders in your catkin workspace to clear the CMake cache and to run `catkin_make`.

UltharPeliki7, 2020-02-19:
I have the same problem across multiple catkin workspaces and have tried both unsourcing and sourcing them, it has also persisted across multiple deletions of devel and build. I have just run ZED_Explorer and the depth viewer and they are both updating at the expected rate(whatever rate is selected in the viewers)

UltharPeliki7, 2020-02-19:
@Myzhar I'm seeing that https://github.com/stereolabs/zed-ros-wrapper/issues/526 has a suspiciously similar frequency rate to mine. I have my zedcam in a separate catkin workspace to the rest of the robot workspace, could the issue be some type of incorrect joint state or URDF configuration?

UltharPeliki7, 2020-02-19:
[URDF files.zip](https://github.com/stereolabs/zed-ros-wrapper/files/4224763/URDF.files.zip)
Here are the two URDF files

Myzhar, 2020-02-19:
If you have modified the default launch files disabling URDF publishing then the cause of your problem can be related to this.
The ZED wrapper requires that the transforms
'zed2_left_camera_frame' -> 'zed2_camera_center` ->  'base_link'
are available.
They are published by the `robot_state_publisher` node if `publish_urdf` is set to true.
If `publish_urdf` is set to false you must provide them

Myzhar, 2020-02-19:
If everything is correct you should read something similar to this on console when the node starts:
```
[ INFO] [1582112008.230706022]: Static transform Sensor to Base [zed_left_camera_frame -> base_link]
[ INFO] [1582112008.230729563]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1582112008.230743748]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1582112008.230769927]: Static transform Sensor to Camera Center [zed_left_camera_frame -> zed_camera_center]
[ INFO] [1582112008.230789802]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1582112008.230805102]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1582112008.230823305]: Static transform Camera Center to Base [zed_camera_center -> base_link]
[ INFO] [1582112008.230837839]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1582112008.230849546]:  * Rotation: {0.000,-0.000,0.000}
```

Myzhar, 2020-02-19:
Otherwise you will get warning messages of this kind after a few seconds:
```[ WARN] [1582112190.920595889]: The tf from 'zed_left_camera_frame' to 'base_link' is not yet available, will assume it as identity!
[ WARN] [1582112190.920637591]: Transform error: "zed_left_camera_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1582112191.422697366]: The tf from 'zed_left_camera_frame' to 'zed_camera_center' is not yet available, will assume it as identity!
[ WARN] [1582112191.422740056]: Transform error: "zed_left_camera_frame" passed to lookupTransform argument target_frame does not exist. 
[ WARN] [1582112191.924213798]: The tf from 'zed_camera_center' to 'base_link' is not yet available, will assume it as identity!
[ WARN] [1582112191.924254625]: Transform error: "zed_camera_center" passed to lookupTransform argument target_frame does not exist. 
```

UltharPeliki7, 2020-02-19:
````
oslaunch zed.launch 
... logging to /home/monarch/.ros/log/679c276c-52e3-11ea-8d3a-79fcc667b840/roslaunch-ASV-17262.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ASV:35689/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.6
 * /zed/zed_description: <?xml version="1....
 * /zed/zed_node/auto_exposure_gain: True
 * /zed/zed_node/auto_whitebalance: True
 * /zed/zed_node/brightness: 4
 * /zed/zed_node/confidence: 97
 * /zed/zed_node/contrast: 4
 * /zed/zed_node/depth/depth_stabilization: 0
 * /zed/zed_node/depth/max_depth: 10.0
 * /zed/zed_node/depth/min_depth: 0.7
 * /zed/zed_node/depth/openni_depth_mode: 0
 * /zed/zed_node/depth/quality: 1
 * /zed/zed_node/depth/sensing_mode: 0
 * /zed/zed_node/exposure: 100
 * /zed/zed_node/gain: 100
 * /zed/zed_node/general/base_frame: base_link
 * /zed/zed_node/general/camera_flip: False
 * /zed/zed_node/general/camera_frame: zed_camera_center
 * /zed/zed_node/general/camera_model: zed
 * /zed/zed_node/general/frame_rate: 2
 * /zed/zed_node/general/gpu_id: -1
 * /zed/zed_node/general/left_camera_frame: zed_left_camera_f...
 * /zed/zed_node/general/left_camera_optical_frame: zed_left_camera_o...
 * /zed/zed_node/general/resolution: 3
 * /zed/zed_node/general/right_camera_frame: zed_right_camera_...
 * /zed/zed_node/general/right_camera_optical_frame: zed_right_camera_...
 * /zed/zed_node/general/self_calib: True
 * /zed/zed_node/general/serial_number: 0
 * /zed/zed_node/general/svo_compression: 0
 * /zed/zed_node/general/verbose: True
 * /zed/zed_node/general/zed_id: 0
 * /zed/zed_node/hue: 0
 * /zed/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed/zed_node/mapping/mapping_enabled: False
 * /zed/zed_node/mapping/resolution: 0
 * /zed/zed_node/mat_resize_factor: 1.0
 * /zed/zed_node/point_cloud_freq: 15.0
 * /zed/zed_node/pos_tracking/fixed_cov_value: 1e-6
 * /zed/zed_node/pos_tracking/fixed_covariance: False
 * /zed/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed/zed_node/pos_tracking/floor_alignment: False
 * /zed/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed/zed_node/pos_tracking/map_frame: map
 * /zed/zed_node/pos_tracking/odometry_db: 
 * /zed/zed_node/pos_tracking/odometry_frame: odom
 * /zed/zed_node/pos_tracking/path_max_count: 10
 * /zed/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed/zed_node/pos_tracking/publish_map_tf: True
 * /zed/zed_node/pos_tracking/publish_pose_covariance: True
 * /zed/zed_node/pos_tracking/publish_tf: True
 * /zed/zed_node/pos_tracking/spatial_memory: False
 * /zed/zed_node/pos_tracking/two_d_mode: False
 * /zed/zed_node/pos_tracking/world_frame: map
 * /zed/zed_node/saturation: 4
 * /zed/zed_node/sharpness: 3
[... 94 lines cut ...]
[ INFO] [1582116047.108828527]: Advertised on topic /zed/zed_node/left_raw/camera_info
[ INFO] [1582116047.126361618]: Advertised on topic /zed/zed_node/right/image_rect_color
[ INFO] [1582116047.126397356]: Advertised on topic /zed/zed_node/right/camera_info
[ INFO] [1582116047.143775061]: Advertised on topic /zed/zed_node/right_raw/image_raw_color
[ INFO] [1582116047.143805778]: Advertised on topic /zed/zed_node/right_raw/camera_info
[ INFO] [1582116047.161038495]: Advertised on topic /zed/zed_node/depth/depth_registered
[ INFO] [1582116047.161071133]: Advertised on topic /zed/zed_node/depth/camera_info
[ INFO] [1582116047.177296039]: Advertised on topic /zed/zed_node/confidence/confidence_image
[ INFO] [1582116047.177346517]: Advertised on topic /zed/zed_node/confidence/camera_info
[ INFO] [1582116047.194288647]: Advertised on topic /zed/zed_node/stereo/image_rect_color
[ INFO] [1582116047.212763088]: Advertised on topic /zed/zed_node/stereo_raw/image_raw_color
[ INFO] [1582116047.213199681]: Advertised on topic /zed/zed_node/confidence/confidence_map
[ INFO] [1582116047.213572073]: Advertised on topic /zed/zed_node/disparity/disparity_image
[ INFO] [1582116047.213985139]: Advertised on topic /zed/zed_node/point_cloud/cloud_registered
[ INFO] [1582116047.214366571]: Advertised on topic /zed/zed_node/pose
[ INFO] [1582116047.214747026]: Advertised on topic /zed/zed_node/pose_with_covariance
[ INFO] [1582116047.215164254]: Advertised on topic /zed/zed_node/odom
[ INFO] [1582116047.215555921]: Advertised on topic /zed/zed_node/path_odom
[ INFO] [1582116047.215943588]: Advertised on topic /zed/zed_node/path_map
[ INFO] [1582116049.228761379]: Static transform Sensor to Camera Center [zed_left_camera_frame -> zed_camera_center]
[ INFO] [1582116049.228858335]:  * Translation: {0.000,-0.060,0.000}
[ INFO] [1582116049.228919407]:  * Rotation: {0.000,-0.000,0.000}
````

This is what I get when I roslaunch zed.launch, I don't get transform errors unfortunately

Myzhar, 2020-02-19:
Elaboration takes 8.03717 sec per frame... there is something weird indeed... 
I saw that you are still on SDK v3.0.1, there is a new patch available, v3.0.2, that fixes a few issues.
Consider also to pull the latest version of the wrapper after you update the SDK

UltharPeliki7, 2020-02-19:
Will do that once I'm back in the lab, time to sleep now.

Myzhar, 2020-02-19:
Meanwhile I suggest you to try to compile the wrapper enabling "debug information":
`$ catkin_make -DCMAKE_BUILD_TYPE=Debug`
in this way you can read on console many useful log messages that could help us to understand where the problem is.
After we find and solve the problem you can go back to "Release" mode simply re-compiling:
`$ catkin_make -DCMAKE_BUILD_TYPE=Release`

vSebas, 2020-02-22:
Could this be related to the transport protocol used in ROS? I understand that the default is TCPROS, would it make a difference if we change it to UDPROS?

Myzhar, 2020-02-22:
No, as we verified also in the issue #526 this problem is often related to a wrong TF tree that does not respect the required ZED TF chain: `<camera_name>_left_camera` -> `<camera_name>__camera_center` -> `base_link`.
@UltharPeliki7 can you please post the file `frames.pdf` generated by the command `$ rosrun tf2_tools view_frames.py` executed while your robotic framework is running

Myzhar, 2020-03-23:
Please close this issue if your problem is solved
@UltharPeliki7

Myzhar, 2020-03-26:
Closed for inactivity

closed 2020-03-26 by Myzhar