stereolabs/zed-ros-wrapper #526: SLOW ZED stream if publish_urdf=false in launch file

federicatrozzi, 2020-02-19:
Hi everyone,
i have a racecar with a zed sensor (SDK3.0), a lidar a nvidia TX2 + other sensors. I'm using ROS melodic + ubuntu 18.04 on my machine and Jetpack 4.2 on nvidia jetson TX2. I thought that was a bandwith issue but thanks to tips of @Myzhar i discovered that my connection bandwidth is wide enought.

I have a "start_sensors.launch" custom file that launches nodes of sensors + the zed one.
If i set parameter <arg name="publish_urdf" value="false" /> the streaming of images topics have the average rate: 0.124 but if i set it true i have the correct average rate (15fps) and the streaming goes fine.
The problem is that if i set it true my localization algorithm does not work (i think is related to a static tf transform).

In fact in my start_sensors. launch i have:

 <param name="robot_description" command="$(find xacro)/xacro --inorder $(arg model)" /> <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" />

and in the zed_camera.launch if i set <arg name="publish_urdf" value="true" /> i have:

<node name="zed_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" output="screen" required="true"> <remap from="robot_description" to="zed_description" /> </node>

i'm investigating on it, but if you have it i appreciate a tip :)

Thank you for your time

federicatrozzi, 2020-02-19:
Update: even if i launch only zed node (so no other sensors nodes are running ) setting `publish_urdf=false` streaming is slow and the average rate is 0.124. So this issue is not related to other sensors i'm using.

Myzhar, 2020-02-19:
A question for you:
have you added the ZED TF tree in your URDF such to publish al the frames required by the ZED?

If you disable the publishing of the ZED TF tree setting `publish_urdf` to false the ZED node internally waits for the frames to be available, failing. That's way is slows down.

federicatrozzi, 2020-02-19:
Oh okay, thank you! So, yes i tried but i don't have changes. I upload files with changes that i made.
I run start_sensors.launch that calls car.urdf.xacro that calls zed.urdf.xacro.  These files have been customized (not from me) so i think there is an error in the zed.urdf.xacro file, but i checked it and seems to be correct! 

[start_sensors.launch.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4225653/start_sensors.launch.txt)
[zed_camera.launch.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4225667/zed_camera.launch.txt)
[car.urdf.xacro.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4225654/car.urdf.xacro.txt)
[zed.urdf.xacro.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4225613/zed.urdf.xacro.txt)

Myzhar, 2020-02-19:
Can you start your system, call `rqt_graph` and post here the generated TF tree.
In this way I can have a more clear idea of what's happening

Myzhar, 2020-02-19:
Sorry, the command for TF tree is this:
`$ rosrun tf2_tools view_frames.py`

If `tf2_tools` is not available, you can install it with
`$ sudo apt install ros-$ROS_DISTRO-rqt-tf-tree`

federicatrozzi, 2020-02-19:
[frames.pdf](https://github.com/stereolabs/zed-ros-wrapper/files/4226247/frames.pdf)

federicatrozzi, 2020-02-19:
sorry that is the right one 
[frames.pdf](https://github.com/stereolabs/zed-ros-wrapper/files/4226264/frames.pdf)

Myzhar, 2020-02-19:
Ok, the problem is finally clear with the TF tree.
The `odom` frame is not linked to `zed_link`, so the transform
map -> odom -> base_link -> zed_link does not exists.

The ZED TF structure is designed to respect the ROS standard according to REP105:
https://www.ros.org/reps/rep-0105.html

Myzhar, 2020-03-23:
Hi @federicatrozzi, please close this issue if your problem is solved

Myzhar, 2020-03-26:
Closed for inactivity

closed 2020-03-26 by Myzhar