stereolabs/zed-ros-wrapper #529: file INSTALL cannot find   "/root/upstream_ws/src/zed-ros-wrapper/zed_wrapper/param".

tianshiz, 2020-02-21:
trying to build tags/v2.8.x on a docker image via ROS Industrial CI + Bitbucket pipelines. I'm pulling this repo as an upstream workspace.
The docker image I'm using is `stereolabs/zed:2.8-ros-cuda9.0-ubuntu16.04`

however I'm getting this error on catkin build:

```
Errors     << zed_wrapper:install /root/upstream_ws/logs/zed_wrapper/build.install.000.log
CMake Error at cmake_install.cmake:248 (file):
  file INSTALL cannot find
  "/root/upstream_ws/src/zed-ros-wrapper/zed_wrapper/param".
make: *** [install] Error 1
cd /root/upstream_ws/build/zed_wrapper; catkin build --get-env zed_wrapper | catkin env -si  /usr/bin/make install; cd -
Failed     << zed_wrapper:install                       [ Exited with code 2 ]
```

I'm not sure what the param file here is. Can I get some help?

databloom, 2020-02-26:
Same issue, directly from github 

-- ==> add_subdirectory(zed-ros-wrapper/zed_wrapper)
CMake Warning at zed-ros-wrapper/zed_wrapper/CMakeLists.txt:22 (find_package):
  By not providing "FindZED.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ZED", but
  CMake did not find one.

  Could not find a package configuration file provided by "ZED" (requested
  version 3) with any of the following names:

    ZEDConfig.cmake
    zed-config.cmake

  Add the installation prefix of "ZED" to CMAKE_PREFIX_PATH or set "ZED_DIR"
  to a directory containing one of the above files.  If "ZED" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at zed-ros-wrapper/zed_wrapper/CMakeLists.txt:17 (message):
  

  

   ZED SDK v3.x not found, install it from:
   https://www.stereolabs.com/developers/ 

  

Call Stack (most recent call first):
  zed-ros-wrapper/zed_wrapper/CMakeLists.txt:23 (checkPackage)

-- Configuring incomplete, errors occurred!
See also "/home/jetbot/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jetbot/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Myzhar, 2020-02-26:
@databloom this is another problem. Here CMake is not finding the ZED SDK. Have you installed the ZED SDK v3.0?
The ZED ROS Wrapper cloned from the master branch requires at least the ZED SDK v3.0.
If you are using SDK v2.8.x you must download the ZED Wrapper from here:
https://github.com/stereolabs/zed-ros-wrapper/releases/tag/v2.x

databloom, 2020-02-26:
Oh you are so great !! I literally just realized I installed it on one but not the other. Thank you for the speedy reply!

closed 2020-03-13 by Myzhar