stereolabs/zed-ros-wrapper #531: The coordinate origin of the ZED camera

eisenWu, 2020-02-24:
Hi, 

i am using the pose data from the ZED camera, i am confused of the coordinata origin of the camera, this is very importante to me. because of the calibration, i have to know where the point exactly is, for example, 2cm in the middle behinde the left camera.
I think the origin of the point cloud is the same to the pose, if i would be wrong, please tell me also where is the origin of the point cloud.

Thanks in advance!!!
Best regards

Myzhar, 2020-02-24:
Hi @eisenWu 
the pose published by the ZED wrapper is automatically referred to the `base_link` frame to make it easier for users to use it and to integrate it in the robotic framework.

The reference frame of the pointcloud is instead in the center of the left camera sensor, 6 cm left of the center of the camera that is exactly in the middle of it.

eisenWu, 2020-02-24:
@Myzhar thanks a lot for your quick answer :)
is that right? the origin of the pose is exactly in the middle of camera, and the origin of the point cloud is 6cm left of the middle?

Myzhar, 2020-02-24:
It's partially correct.
That's true only if `zed_camera_frame` and `base_link` are cohincident.
If you modify the launch file here
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed.launch#L29-L34
you introduce an offset between `zed_camera_frame` and `base_link` and the pose will be referred to `base_link`.
For more info:
https://www.ros.org/reps/rep-0105.html

eisenWu, 2020-02-24:
@Myzhar Thank you!!!

closed 2020-02-24 by eisenWu