stereolabs/zed-ros-wrapper #533: imu cam Transform

wangq0688, 2020-02-28:
hi,

Camera-IMU Transform: 
 7FFF66E12360
0.999821 0.000465 -0.018939 -0.002000
-0.000232 0.999925 0.012280 -0.015000
0.018943 -0.012273 0.999745 -0.002000
0.000000 0.000000 0.000000 1.000000

I found it means IMU to Left camera transform matrix, that contains rotation and translation between IMU frame and camera frame.
How can i get IMU to Right cam transform matrix?

thanks

Myzhar, 2020-02-28:
Hi @wangq0688,
you cannot get it directly, but you can calculate it simply inverting the matrix that you posted and multiplying it for the roto-translation matrix from left camera to right camera:
```
[ 1.0 0   0    0      ]
[ 0   1.0 0    -0.12  ]
[ 0   0   1.0  0      ]
[ 0   0   0    0      ]
```

wangq0688, 2020-02-28:
thanks for your speedy reply.
Camera-IMU Transform means IMU to Left camera transform matrix. I'm not sure if I reversed it.

wangq0688, 2020-03-05:
hi
topic /zedm/zed_node/left_raw/camera_info    
   
` frame_id: "zedm_left_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [-0.17619900405406952, 0.029474100098013878, 0.0, 0.0008147479966282845, -0.000478577014291659]
K: [700.135009765625, 0.0, 639.4099731445312, 0.0, 700.135009765625, 350.16900634765625, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [700.135009765625, 0.0, 639.4099731445312, 0.0, 0.0, 700.135009765625, 350.16900634765625, 0.0, 0.0, 0.0, 1.0, 0.0]
`

and on topic /zedm/zed_node/right_raw/camera_info

`P: [697.760009765625, 0.0, 643.0349731445312, -44.12062072753906, 0.0, 697.760009765625, 347.31451416015625, 0.0, 0.0, 0.0, 1.0, 0.0]
`

Calculated rotation and translation matrix from left camera to right camera:
`[ 1.003403749,            0,                     -0.005195195,                0.0632317990]
[ 0,                                1.00340374,    0.0040909369,                        0]
[ 0,                                0,                          1.0,                                   0]
[0,                                 0,                            0,                                 1.0]`

I find Baseline=63.0173     
Is this the correct rotation translation matrix?

Myzhar, 2020-03-05:
@wangq0688 it is

closed 2020-03-05 by wangq0688