stereolabs/zed-ros-wrapper #537: Significance of `CuCtxGetDeviceId` inside ZEDRosWrapperNodelet unclear

kvsaivineeth, 2020-03-12:
Hi,

Raising this issue to understand what is the real significance of the variable of the CuCtxGetDeviceId or how `mGpuId` can be utilized in the GPU based application development using Nodelets. More documentation on this would really be appreciated. Also, any leads on how to make use of CUDA pointers inside the publish functions in the wrapper nodelet to access GPU resources (i.e, data published from GPU from the current CPU method)

Myzhar, 2020-03-12:
Hi @kvsaivineeth,
the `CuCtxGetDevice` command is used to get the ID of the GPU on which the ZED SDK is running.
It is useful if you have a system with multiple Nvidia cards and you want to know which of them is working.
By default the ZED SDK runs on the most powerful GPU present in the system, but you can force it to use the GPU you prefer using the `general/gpu_id` parameter (see `common.yaml`).

I guess that there are no ways to use "CUDA pointers" inside the publish functions in ROS. ROS is not really CUDA friendly, you can use CUDA inside a node or a nodelet, but as far as I know it is not possible to share GPU memory between processes or inside the same process using nodelets.

It could be very useful if it was possible because the ZED SDK allows to get access to image and depth data directly on GPU memory, but to publish them in the ROS world we must always transfer them to CPU memory.

kvsaivineeth, 2020-03-12:
Thank you @Myzhar for the quick response, appreciate your time.

closed 2020-03-12 by kvsaivineeth