stereolabs/zed-ros-wrapper #538: Unable to receive original imu message (ros_wrapper) when sensors_timestamp_sync is true

arenas7307979, 2020-03-13:
hi, 
I tried to adjust the parameters of the zed2 camera, and have few questions
1. Unable to receive original imu message (ros_wrapper) when sensors_timestamp_sync is true
2. Is there any tool to support calibration of of imu-camera extrinsic parameter and rolling shutter time, I've tried the kalibr tool before, but there were some problems for rolling shutter time

Myzhar, 2020-03-13:
Hi @arenas7307979 
1. Can you explain better your problem?
2. Kalibr is the best tool for this kind of calibration. Check your configuration and be sure to make the right movements while recording the video. Have you read our tutorial? https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-

arenas7307979, 2020-03-15:
hii, about first problem 
I tried to set the sensors_timestamp_sync "true" of the zed2 camera, didn't get any messages from the imu

[image]

[image]

set the sensors_timestamp_sync "false", Imu data can be received. 

[image]

Myzhar, 2020-03-15:
If you enable the synchronization with the images you must subscribe also to an image topic otherwise there is nothing to synchronize.
Open another console and type 'rostopic hz /zed2/zed_node/left/image_rect_color', you will see that the IMU data starts to be published at the same frequency of the image topic

arenas7307979, 2020-03-16:
I found the reason, thank you
I did not subscribe the topic of image_rect_color  
because I used the original image (for fov)

Myzhar, 2020-03-23:
Hi @arenas7307979 if your problem is solved please close this issue

closed 2020-03-24 by arenas7307979