stereolabs/zed-ros-wrapper #539: fatal error LNK1120: 1 unresolved externals - Building zed-ros-wrapper on ROS On Windows

uchuuengineer, 2020-03-13:
Hi, I am unable to build the latest release of zed-ros-wrapper for my ROS build. I am running ROS Melodic on Windows and I was previously successful in building the zed-ros-wrapper approximately a month ago, back when I think the most recent ZED SDK was 3.0.2. But after updating my SDK to 3.1, my old zed-ros-wrapper build stopped being able to detect my camera and the new wrapper fails to build for me. 

Here is the error message I am getting while trying to build the current zed-ros-wrapper with the 3.1 SDK. Please let me know if there is anything I can do to correct this error: 

```
c:\ws\zed2>catkin_make
Base path: c:\ws\zed2
Source space: c:\ws\zed2\src
Build space: c:\ws\zed2\build
Devel space: c:\ws\zed2\devel
Install space: c:\ws\zed2\install
####
#### Running command: "cmake c:\ws\zed2\src -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCATKIN_DEVEL_PREFIX=c:\ws\zed2\devel -DCMAKE_INSTALL_PREFIX=c:\ws\zed2\install -G NMake Makefiles" in "c:\ws\zed2\build"
####
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is MSVC 19.24.28319.0
-- The CXX compiler identification is MSVC 19.24.28319.0
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
-- Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe
-- Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.24.28314/bin/Hostx64/x64/cl.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: C:/ws/zed2/devel
-- Using CMAKE_PREFIX_PATH: C:/opt/ros/melodic/x64;C:/opt/rosdeps/x64;C:/opt/vcpkg/installed/x64-windows
-- This workspace overlays: C:/opt/ros/melodic/x64
-- Found PythonInterp: C:/opt/python27amd64/python.exe (found suitable version "2.7.15", minimum required is "2")
-- Using PYTHON_EXECUTABLE: C:/opt/python27amd64/python.exe
-- Using default Python package layout
-- Found PY_em: C:\opt\python27amd64\lib\site-packages\em.pyc
-- Using empy: C:/opt/python27amd64/lib/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: C:/ws/zed2/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: C:/opt/python27amd64/Scripts/nosetests-2.7.exe
-- catkin 0.7.21
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 3 packages in topological order:
-- ~~  - zed_interfaces
-- ~~  - zed_ros (metapackage)
-- ~~  - zed_wrapper
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'zed_interfaces'
-- ==> add_subdirectory(zed-ros-wrapper/zed_interfaces)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- zed_interfaces: 2 messages, 13 services
-- +++ processing catkin metapackage: 'zed_ros'
-- ==> add_subdirectory(zed-ros-wrapper/zed_ros)
-- +++ processing catkin package: 'zed_wrapper'
[... 377 lines cut ...]
            M=sensor_msgs::CameraInfo
        ]
C:\ws\zed2\src\zed-ros-wrapper\zed_wrapper\src\nodelet\src\zed_wrapper_nodelet.cpp(1902): note: see reference to function template instantiation 'void ros::Publisher::publish<sensor_msgs::CameraInfo>(const boost::shared_ptr<sensor_msgs::CameraInfo> &) const' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(54): note: see reference to class template instantiation 'boost::arg<9>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(53): note: see reference to class template instantiation 'boost::arg<8>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(52): note: see reference to class template instantiation 'boost::arg<7>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(51): note: see reference to class template instantiation 'boost::arg<6>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(50): note: see reference to class template instantiation 'boost::arg<5>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(49): note: see reference to class template instantiation 'boost::arg<4>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(48): note: see reference to class template instantiation 'boost::arg<3>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(47): note: see reference to class template instantiation 'boost::arg<2>' being compiled
C:\opt\rosdeps\x64\include\boost-1_66\boost/bind/placeholders.hpp(46): note: see reference to class template instantiation 'boost::arg<1>' being compiled
[ 97%] Linking CXX shared library C:\ws\zed2\devel\bin\ZEDWrapper.dll
LINK Pass 1: command "C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1424~1.283\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\ZEDWrapper.dir\objects1.rsp /out:C:\ws\zed2\devel\bin\ZEDWrapper.dll /implib:C:\ws\zed2\devel\lib\ZEDWrapper.lib /pdb:C:\ws\zed2\devel\bin\ZEDWrapper.pdb /dll /version:0.0 /machine:x64 /debug /INCREMENTAL /DEF:CMakeFiles\ZEDWrapper.dir\exports.def -LIBPATH:C:\PROGRA~2\ZEDSDK~1\lib C:\opt\ros\melodic\x64\lib\image_transport.lib C:\opt\ros\melodic\x64\lib\dynamic_reconfigure_config_init_mutex.lib C:\opt\ros\melodic\x64\lib\nodeletlib.lib C:\opt\ros\melodic\x64\lib\bondcpp.lib C:\Program Files (x86)\Windows Kits\10\Lib\10.0.18362.0\um\x64\RpcRT4.Lib C:\opt\ros\melodic\x64\lib\class_loader.lib C:\opt\rosdeps\x64\lib\PocoFoundation.lib C:\opt\ros\melodic\x64\lib\roslib.lib C:\opt\ros\melodic\x64\lib\rospack.lib C:\opt\python27amd64\libs\python27.lib C:\opt\rosdeps\x64\lib\boost_program_options-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\tinyxml2.lib C:\opt\ros\melodic\x64\lib\orocos-kdl.lib C:\opt\ros\melodic\x64\lib\tf2_ros.lib C:\opt\ros\melodic\x64\lib\actionlib.lib C:\opt\ros\melodic\x64\lib\message_filters.lib C:\opt\ros\melodic\x64\lib\tf2.lib C:\opt\ros\melodic\x64\lib\roscpp.lib C:\opt\rosdeps\x64\lib\boost_filesystem-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\rosconsole.lib C:\opt\ros\melodic\x64\lib\rosconsole_log4cxx.lib C:\opt\ros\melodic\x64\lib\rosconsole_backend_interface.lib C:\opt\rosdeps\x64\lib\log4cxx.lib C:\opt\rosdeps\x64\lib\boost_regex-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\xmlrpcpp.lib C:\opt\ros\melodic\x64\lib\roscpp_serialization.lib C:\opt\ros\melodic\x64\lib\rostime.lib C:\opt\ros\melodic\x64\lib\cpp_common.lib C:\opt\rosdeps\x64\lib\boost_system-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_thread-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_chrono-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_date_time-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\boost_atomic-vc141-mt-x64-1_66.lib C:\opt\rosdeps\x64\lib\console_bridge.lib C:\Program Files (x86)\ZED SDK\lib\sl_zed64.lib C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2\lib\x64\cudart_static.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\ZEDWrapper.dir/intermediate.manifest CMakeFiles\ZEDWrapper.dir/manifest.res" failed (exit code 1120) with the following output:
   Creating library C:\ws\zed2\devel\lib\ZEDWrapper.lib and object C:\ws\zed2\devel\lib\ZEDWrapper.exp
zed_wrapper_nodelet.cpp.obj : error LNK2019: unresolved external symbol cuCtxGetDevice referenced in function "protected: virtual void __cdecl zed_wrapper::ZEDWrapperNodelet::onInit(void)" (?onInit@ZEDWrapperNodelet@zed_wrapper@@MEAAXXZ)
[... cut ...] nals
NMAKE : fatal error U1077: 'C:\opt\python27amd64\Lib\site-packages\cmake\data\bin\cmake.exe' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.24.28314\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed
```

Myzhar, 2020-03-13:
Hi @uchuuengineer 
`cuCtxGetDevice` is part of CUDA. The link error means that the linker cannot find CUDA will linking to generate  ZEDWrapper.dll 
We have never tested the ZED Wrapper under Windows so we cannot provide a direct solution to this kind of error.
What I can suggest you is to clean the CMake cache and to try to recompile

uchuuengineer, 2020-03-17:
Hi Myzhar,

I was able to re-clone and built the January 28 update of zed-ros-wrapper. It appears version [8bbd6693cde8aaffc34d9af64835bc2544e03f41](https://github.com/stereolabs/zed-ros-wrapper/commit/8bbd6693cde8aaffc34d9af64835bc2544e03f41) builds on Windows, while most of the others do not. I will just be using that version for now. Thanks.

closed 2020-06-23 by Myzhar