stereolabs/zed-ros-wrapper #540: Object detection running but no person detected

ThomasJ08, 2020-03-16:
Hi,
I am trying to implement person detection and tracking using a Zed 2 camera, a Jetson AGX Xavier, the Zed SDK and the zed_ros_wrapper. I managed to make it work with version 3.0.1 of the SDK but since I upgraded to version 3.1 the object detection starts but no message is being published on the /zed2/zed_node/obj_det/object topic even though there is a person in the camera field of view.
I already tried to reinstall the last version of the SDK and download the object model at first start of the object detection module. I tried to reduce the detection threshold but with no result.
Is there a modification in the new SDK version that might make object detection incompatible with the zed_ros_wrapper ? Am I doing something wrong or did I miss something ?
Thank you in advance for your help.
Best regards,
Thomas

Myzhar, 2020-03-16:
Hi @ThomasJ08 
can you please try the Object Detection sample in 
`/usr/local/zed/samples/object detection/cpp`
to be sure that it works correcty?

ThomasJ08, 2020-03-16:
It is working correctly with this sample even though I get the following error when launching it :
Gtk-Message: 16:35:34.362: Failed to load module "canberra-gtk-module"

Myzhar, 2020-03-16:
I'm going to investigate which can be the problem. Stay tuned

Myzhar, 2020-03-17:
@ThomasJ08 the issue has been reproduced. We are on it.

Myzhar, 2020-03-17:
We found a bug in ZED SDK v3.1 that doesn't allow to correctly perform Object Tracking when the coordinate system is set to `sl::COORDINATE_SYSTEM::RIGHT_HANDED_Z_UP_X_FWD;` (this is the ROS case).

We are working on a patch to fix it, meanwhile to perform Object Detection in ROS I suggest to use the SDK v3.0.3.

ThomasJ08, 2020-03-17:
Ok, I will go back to SDK v3.0.3 while you fix this bug. 
Please keep me updated when this bug is fixed.
Thank you.

Myzhar, 2020-03-18:
The bug has been fixed at SDK level.
Please follow our channels to be informed when the SDK patch will be released.

Website: https://www.stereolabs.com/developers/release/
Twitter: https://twitter.com/Stereolabs3D

@ThomasJ08 thank you very much for reporting the bug :+1:

closed 2020-03-18 by Myzhar

Myzhar, 2020-03-20:
@ThomasJ08 the new SDK v3.1.1 with the fix for this bug has been released