stereolabs/zed-ros-wrapper #541: zed_camera_nodelet.launch with Zed2

jsseng, 2020-03-18:
I am trying to launch this nodelet launch file, but I am getting an error when it is looking for a .urdf file.  In the launch file, it only list the 'zed' and 'zedm' as camera models.

The error is:

IOError: [Errno 2] No such file or directory: u'/home/jseng/h2_ws/src/zed-ros-wrapper/zed_wrapper/urdf/zed.urdf'

Myzhar, 2020-03-20:
@jsseng are you using the latest version of the wrapper?

jsseng, 2020-03-20:
I just updated to ver. 3.1.1 of the SDK and the latest version of the wrapper.  It is running now (I haven't check the images yet), but it seems to publish all the topics under the /zed namespace instead of /zed2.  I tried modifying zed_camera_nodelet.launch in the wrapper, but it still publishes under the /zed namespace.

Myzhar, 2020-03-20:
The namespace is defined by the `camera_name` parameter, not by the `camera_model`

Myzhar, 2020-03-20:
@jsseng you can read more on the new ZED Nodelet documentation:
https://www.stereolabs.com/docs/ros/zed_nodelets/

jsseng, 2020-03-20:
Okay, great.  It seems to be working now.  Thanks!

closed 2020-03-20 by jsseng