stereolabs/zed-ros-wrapper #544: Installation error when invoking rosdep

BraedenCullenCodes, 2020-03-20:
After running the following: rosdep install --from-paths src --ignore-src -r -y

I receive the following error : 
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_examples: Cannot locate rosdep definition for [zed_display_rviz]
zed_obj_det_sub_tutorial: Cannot locate rosdep definition for [zed_wrapper]
zed_video_sub_tutorial: Cannot locate rosdep definition for [zed_wrapper]
zed_tracking_sub_tutorial: Cannot locate rosdep definition for [zed_wrapper]
zed_depth_sub_tutorial: Cannot locate rosdep definition for [zed_wrapper]
zed_sensors_sub_tutorial: Cannot locate rosdep definition for [zed_wrapper]
zed_nodelet_example: Cannot locate rosdep definition for [zed_wrapper]
zed_wrapper: Cannot locate rosdep definition for [zed_interfaces]
zed_multicamera_example: Cannot locate rosdep definition for [zed_wrapper]
zed_ar_track_alvar_example: Cannot locate rosdep definition for [zed_wrapper]
zed_display_rviz: Cannot locate rosdep definition for [zed_wrapper]
zed_ros: Cannot locate rosdep definition for [zed_wrapper]
zed_rtabmap_example: Cannot locate rosdep definition for [zed_wrapper]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

(I am on Jetson tx2 with most recent ZED SDK)

Myzhar, 2020-03-20:
Hi @BraedenCullenCodes 
you are getting this error, but everything is ok and you can continue to install the wrapper.
Go to the root of your catkin workspace and run `catkin_make`.
The compiling process will be successful.

You are getting this error because you installed cloned the zed-ros-examples repository before compiling the zed-ros-wrapper repository

closed 2020-03-20 by BraedenCullenCodes

Baldins, 2021-03-19:
Same problem here but I also get error when runninc catkin_make. 
Any chance you know how to fix it?

Thank you!

Myzhar, 2021-03-19:
@Baldins If you compile the examples the first time when the `zed_wrapper` has not yet been compiled `rosdep` cannot find it and the dependency cannot be satisfied.
Like I wrote above this is not a blocking problem because `catkin_make` should work.
If it generates an error there is something other that is not correct.
Please open a new issue reporting all the useful information to help soviet your problem, mainly copy and paste the error itself

Baldins, 2021-03-19:
That's the problem - catkin_make is not working in my case - It gives me zed interface Config errors

maximusron, 2021-09-20:
> That's the problem - catkin_make is not working in my case - It gives me zed interface Config errors

Have you found a fix? i am running into the same error

Myzhar, 2021-09-20:
@maximusron what error? Can you please copy and paste the compile log in order to provide more information?

maximusron, 2021-09-20:
> @maximusron what error? Can you please copy and paste the compile log in order to provide more information?

on running  rosdep install --from-paths src    in the ws :
                                                
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
zed_ros: Cannot locate rosdep definition for [zed_wrapper]
zed_nodelets: Cannot locate rosdep definition for [zed_interfaces]
zed_wrapper: Cannot locate rosdep definition for [zed_nodelets]

maximusron, 2021-09-20:
> @maximusron what error? Can you please copy and paste the compile log in order to provide more information?
it also gives a lot of errors while catkin making it
[image]
[image]

Myzhar, 2021-09-20:
Hi @maximusron 
as you can see this is a totally different issue. The compiler correctly founds the `zed_interfaces` package, but it cannot compile it because you are using a NOT COMPATIBLE version of the ZED SDK.
The ZED ROS Wrapper in the main branch is designed to support the latest version of the ZED SDK with all the new available features.
What version of the ZED SDK are you using?

PS please never reply to a CLOSED ISSUE, it's always better to create a NEW ISSUE and add a link to the closed one to cite it.

AndreV84, 2022-04-07:
@Myzhar I have renewed the issue at https://github.com/stereolabs/zed-ros-wrapper/issues/829
Do I stil need to write to email to get response regarding the issue or it wil be addressed at github somehow?

Praveen290192, 2022-04-11:
I had the same issue but was able to resolve it.
When we clone without recursive we get those issues at least that was my case. When I cloned it using recursive it build without any issues.
[link](https://github.com/stereolabs/zed-ros2-wrapper#:~:text=git%20clone%20%20%2D%2Drecursive%20https%3A//github.com/stereolabs/zed%2Dros2%2Dwrapper.git)