stereolabs/zed-ros-wrapper #546: Extending the ZEDWrapperNodelet

kvsaivineeth, 2020-03-21:
Hi, I am currently trying to extend the supplied ZEDWrapperNodelet into a custom nodelet of my requirements. For the same, i have been writing something similar to this :
My question is how to write the `nodelet_plugin.xml` file and what is the base class i need to export to in the nodelet/src file? Also wanted to know if this segment also overrides the default ZEDWrapperNodelet's onInit() . 

```<library path="lib/libmyNodelet">
  <class name="ns/myNodelet" type="ns::myNodelet" base_class_type="nodelet::Nodelet">
  <description>
  This is my nodelet.
  </description>
  </class>
</library>
```
``` 
namespace ns{
                class myNodelet:ZEDWrapperNodelet {
                 public:
                        myNodelet::myNodelet():ZEDWrapperNodelet(){}
                        myNodelet:~myNodelet(){}
                         virtual void onInit(){ 
                                       m=getMTNodeHandle();
                                       NODELET_INFO("Initialized nodelet");
                                       sub_=m.subscribe("/topic",10,&myNodelet::callback,this);
                               }
                 protected:
                          virtual callback(const sensor_msgs::ImagePtr &msg){
                           NODELET_INFO("Subscribing to Image topic");
                             }            
```
#include<pluginlib/class_list_macro.h>
PLUGINLIB_EXPORT_CLASS(ns:myNodelet,nodelet::Nodelet / ns:ZEDWrapperNodelet);

Myzhar, 2020-03-23:
Hi @kvsaivineeth 
in your case you must write a `nodelet_plugins.xml` similar to:
```
<library path="lib/libMyNodelet"> <!-- use the name of you lib exported in CMakeFiles.txt -->
    <class name="ns/myNodelet"
           type="ns::myNodelet"
           base_class_type="zed_nodelets::ZEDWrapperNodelet">
      <description>Add your description here</description>
    </class>
</library>
```

**Note**: the code you pasted above is better if you replace 
```
class myNodelet:ZEDWrapperNodelet {
```
with
```
class myNodelet : public zed_nodlets::ZEDWrapperNodelet {
```

This is a list of useful guides to write nodelets:
- Clearpath: https://www.clearpathrobotics.com/assets/guides/ros/Nodelet%20Everything.html
- Medium: https://medium.com/@waleedmansoor/understanding-ros-nodelets-c43a11c8169e

It is also very important to understand how a plugin works:
http://wiki.ros.org/pluginlib

Let me know it this helps and do not hesitate to ask for more info

kvsaivineeth, 2020-03-23:
Thank you for the detailed information on this @Myzhar , appreciate it very much. 

Alongwith my previous question, also wanted to understand if the zed_nodelets package should co-exist with my catkin packages or what is the best way to access the zed_nodelets according to you ?

Myzhar, 2020-03-24:
If you use different packages compiled from source you must put them in the same catkin workspaces.
In this particular case if you don't put your new package in the same catkin workspace of the zed_nodelets package you will not be able to see the ZEDWrapperNodelet base class for your nodelet

kvsaivineeth, 2020-04-01:
Hi @Myzhar , Thanks for the guidance in understanding the ZED Nodelet. I was trying to understand the significance of mPcThread & mDevicePollThread used inside the Nodelet. What would be the advantage of having different threads in a nodelet interms of efficiency. Also, could we use a main function like Nodes and how does this compare with using threads ?

closed 2020-05-06 by kvsaivineeth