stereolabs/zed-ros-wrapper #547: saveAreaMap issue

oyaremchuk, 2020-03-24:
Hi,
I'm trying to save area map. For this I have modified zed_wrapper_nodelet.cpp
Unfortunately, sl::ERROR_CODE err = mZed.saveAreaMap returned FAILURE, but map.txt created without issue.
Could you advise me, how to fix issue and what is wrong?
Configuration: Zedm, SDK 3.1.1, Jetson TX2 4.3, ROS Melodic

```
std::ofstream file("/home/jetson/projects/area/map.txt", ios::out | ios::binary);
file << "Writing this to a file.\n";
file.close();

sl::ERROR_CODE err = mZed.saveAreaMap("/home/jetson/projects/area/map.area");
std::cout << "saveAreaMap" << std::endl;
std::cout << err << std::endl;
for (int i = 0; i < 100; i++)
{
    sl::AREA_EXPORTING_STATE state = mZed.getAreaExportState();
    std::cout << i << " " << state << std::endl;
    ros::Duration(0.05).sleep();
}
mZed.close(); 
```

Myzhar, 2020-03-25:
Hi @oyaremchuk 
I test it and let you know.

oyaremchuk, 2020-03-31:
Hi @Myzhar 
I have tried simple node with zed position tracing and saveAreaMap works for it.
[test_node.cpp.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4411605/test_node.cpp.txt)

Do you know what is wrong with zed_wrapper_nodelet.cpp and why saveAreaMap returned FAILURE?

Thanks for help!
P.S. Do you have plan to create ros service which run saveAreaMap? I have drafts and can help with it.

Myzhar, 2020-03-31:
There are many causes for failure:
https://www.stereolabs.com/docs/api/classsl_1_1Camera.html#afe82ca1a5a374233a0f28bf7ba747674

A service to issue an area map saving will be added soon 👍

Myzhar, 2021-07-28:
Finalled added with #766

closed 2021-07-28 by Myzhar