stereolabs/zed-ros-wrapper #548: ZED connection -> CAMERA FAILED TO SETUP

ruksen25, 2020-03-25:
Hi!

I am working on a project that I have build up inside a docker-container. 
The cotainer is set up as follows:

- Ubuntu 16.04
-  Cuda 10.2 & cudnn7
- Ros Kinetic
- ZED SDK 3.1
And i use the ZED2. 

My container has GL and so I am able to open ZED_Explorer tool which shows the image feed from the camera just fine, but when running it gives the following error:
`[ZED Depth Viewer] Failed to create directory to save images: /root/Documents/ZED
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
`

 When I run the ros wrapper with `roslaunch zed_wrapper zed2.launch` I get similar errors regarding camera setup and sl::Camera::Open not being run. 
Do you know what might cause this issue?

This is the full output of running the ROS wrapper:

`$ roslaunch zed_wrapper zed2.launch                     
... logging to /root/.ros/log/942b1d46-6e78-11ea-9010-0242ac110002/roslaunch-1d22dc2e1326-10293.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://1d22dc2e1326:34895/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: 0
 * /zed2/zed_node/depth/quality: 3
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
[... 95 lines cut ...]
[ INFO] [1585127477.761893657]:  * confidence_frame		-> zed2_left_camera_frame
[ INFO] [1585127477.761991779]:  * confidence_optical_frame	-> zed2_left_camera_optical_frame
[ INFO] [1585127477.764576918]:  * Broadcast odometry TF	-> ENABLED
[ INFO] [1585127477.766959156]:  * Broadcast map pose TF	-> ENABLED
[ INFO] [1585127477.768142751]:  * [DYN] Depth confidence	-> 50
[ INFO] [1585127477.769491331]:  * [DYN] Depth texture conf.	-> 100
[ INFO] [1585127477.770741613]:  * [DYN] pub_frame_rate		-> 15 Hz
[ INFO] [1585127477.772087303]:  * [DYN] point_cloud_freq	-> 15 Hz
[ INFO] [1585127477.773334782]:  * [DYN] brightness		-> 4
[ INFO] [1585127477.774660803]:  * [DYN] contrast		-> 4
[ INFO] [1585127477.776064827]:  * [DYN] hue			-> 0
[ INFO] [1585127477.777620750]:  * [DYN] saturation		-> 4
[ INFO] [1585127477.778926578]:  * [DYN] sharpness		-> 4
[ INFO] [1585127477.780301103]:  * [DYN] gamma			-> 8
[ INFO] [1585127477.781399724]:  * [DYN] auto_exposure_gain	-> ENABLED
[ INFO] [1585127477.785017651]:  * [DYN] auto_whitebalance	-> ENABLED
[ INFO] [1585127477.802464788]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1585127477.802624385]:  *** Opening ZED 2...
[ INFO] [1585127479.239753542]: ZED connection -> CAMERA FAILED TO SETUP
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585127482.767643286]: ZED connection -> CAMERA FAILED TO SETUP
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585127486.175643044]: ZED connection -> CAMERA FAILED TO SETUP
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
`

adujardin, 2020-03-25:
Hi

What command did you use to start the container? Did you use `--privileged` option?  
Did you plug the ZED2 after launching the container (see https://github.com/stereolabs/zed-docker#usb-replughot-plug if so)?

Could you launch `lsusb -d 2b03: ` from within the container (you may need to install `usbutils`)?

If possible could you share the docker image or DockerFile used and the exact launch command to try to replicate this?

ruksen25, 2020-03-25:
Thank you for the reply. I really appreciate you taking the time to offer support.

I run using this command:
`sudo docker run --runtime=nvidia -v /home/usr/Documents/ergo_adapt_docker_share/ergo_adapt_docker_ros:/home/catkin_ws/src/ergoadapt -it --rm --name zed --device=/dev/video0:/dev/video0 --device=/dev/video1:/dev/video1 --privileged  --add-host=021802CP00066.local:192.168.1.19 -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro mkvdb/ros_intera_zed_openpose:latest bash`

Using the last part (image name) you should be able to download my image from docker hub.

Camera is plugged in before running the container.
When running `lsusb -d 2b03:` in my container I get:
`Bus 001 Device 006: ID 2b03:f781  
Bus 001 Device 005: ID 2b03:f780 `

adujardin, 2020-03-25:
Your lsusb command shows the ZED2 is correctly detected (Camera and IMU). 

I tried different things with zed images and your image and it's working fine for both ZED and ZED2.

I did not use this exact launch command this my nvidia docker version is newer and I don't have some of the local folders.

`docker run --gpus all -e NVIDIA_DRIVER_CAPABILITIES=all -it --rm --name zed --privileged -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:ro mkvdb/ros_intera_zed_openpose:latest bash`

[Output logs](https://github.com/stereolabs/zed-ros-wrapper/files/4380440/logs.txt)

I also tried the tools (you can ignore the Depth viewer error about the folder creation, it's working fine without it) and some samples without any issues.

Could you try running a sample without ROS on a ZED image ([DockerHub](https://hub.docker.com/r/stereolabs/zed/), [zed-docker](https://github.com/stereolabs/zed-docker)) ?

ruksen25, 2020-03-25:
Your output log file indeed seems to indicate that you do not experience the same issues. 
When I run my container with your docker run command nothing changes and I still have the same issues. 

I tried running the stereolabs/zed:3.1-gl-devel-cuda10.0-ubuntu16.04 container (which resembles my setup the most i think) and then launch the ZED_Depth_viewer inside. When I do that, I still get the `[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
` error. 
I have previously tested in several of your different docker containers but my error occurs in each of them.

I find it all very peculiar. 

I am using a brand new camera for the very first time. Is there any initialization of the camera I habe to do which I have overlooked? 
Also What versions of Docker, its API and the Nvidia-container-toolkit do you have? (Grasping at straws here...) 

Inside the mentioned container of yours I tested, I should be able to cmake inside eg. /usr/local/zed/samples/depth sensing/cpp/ , right?

adujardin, 2020-03-25:
I'm using Ubuntu 18, Cuda 10.0 with nvidia-container-toolkit (not nvidia-docker2). But I think this would only result in an issue with CUDA execution or GPU related issue (video decoding, OpenGL...), which seems to be working fine here.

No, there's nothing to initialize or any special first step for new cameras. 
Depending on the docker image used, but on most on them, you can directly compile (cmake/make) the sample in interactive yes.

Would it be possible for you to test the tools or samples directly on the host? 
Can you run `guvcview` in docker to try to open the camera without the ZED SDK? Maybe also the Diagnostic Tool, and upload the json results.

adujardin, 2020-03-25:
One other thing, are you sure the ZED2 is connected to a USB3 port? The error code you have usually means the resolution can't be set because there's not enough bandwidth.

ruksen25, 2020-03-26:
This is a bit embarrassing, but it seems like your last comment was helpful. 

Though I had been aware of the requirement of a USB3, It seems like I had failed to ensure this is what I am using. Since my visualisation in the SDK tools was working just fine, I assumed it was correctly connected. 
However, I had connected my camera to a usb port in the front of my desktop which has a blue insert blindly trusting it had the sufficient USB3 capeabilities. Seemingly this was not the case. I then connected the camera on the back of the tower and now it works. [Apparently this is a know issue.](https://support.stereolabs.com/hc/en-us/articles/207635225-How-to-fix-USB-3-0-bandwidth-and-connection-issues)
For anyone else reading this wanting to ensure they are using the right USB, I can share that I found it helpful to run ` lsusb -d 2b03: -v` and then inspect the bcdUSB value for the camera to check that it is indeed 3.00

My ROS zed_wrapper is now running flawlessly, and I am able to visualise in rviz. 

BTW. I found out that it also works if I take out the `--privileged` flag and and add the two buses obtained by `lsusb -d 2b03:` and video0 and video1, thus in my case using the flags `--device=/dev/video1:/dev/video1 --device=/dev/video0:/dev/video0 --device=/dev/bus/usb/004:/dev/bus/usb/004 --device=/dev/bus/usb/003:/dev/bus/usb/003`. When I have alternate options I prefer that to the `--privileged` flag that you seem to recommend most frequently.

@adujardin Thank you very much for your support it has been a huge help.

closed 2020-03-26 by ruksen25

adujardin, 2020-03-26:
The documentation is mainly focused on getting started. We recommend the `privileged` option to avoid having to check which dev folder to mount for each camera, and ease of development in general, but that's correct it is not mandatory.