stereolabs/zed-ros-wrapper #555: [Question] When I run zed2.launch, it is too slow.

Kimsuhyung, 2020-04-24:
When I clone ros_wrapper and run zed2.launch, if feels like an image comes in every second (abount 1FPS).

But, I confirmed the image came out well through the example (/usr/local/zed/tools)

So, I tried changing the resolution and frame rate. Bunt there is no change.

I am using nvidia xavier and The power mode was MAXN.

Also, I installed the latest sdk 3.1.2

Does anyone have the same problem as me?

If anyone knows how to fix it, please let me know.

Thank you !

Myzhar, 2020-04-24:
Hi @Kimsuhyung 
how are you testing the framerate?

Can you try this?
1. Open a console and  `$ roslaunch zed_wrapper zed2.launch`
2. Open a second console and `$ rostopic hz /zed2/zed_node/left/image_rect_color`

Let me know the output

G-Pereira, 2020-05-08:
Hello @Myzhar and @Kimsuhyung , I am getting exactly the same "error" and I suspect I am setting the wrong GPU ID (-1 is default). What should it be for NVIDIA Jetson TX2?

> Hi @Kimsuhyung
> how are you testing the framerate?
> 
> Can you try this?
> 
>     1. Open a console and  `$ roslaunch zed_wrapper zed2.launch`
> 
>     2. Open a second console and `$ rostopic hz /zed2/zed_node/left/image_rect_color`
> 
> 
> Let me know the output

I get this output from those two commands:

subscribed to [/zed2/zed_node/left/image_rect_color]
no new messages
no new messages
no new messages
no new messages
average rate: 0.204
	min: 4.897s max: 4.897s std dev: 0.00000s window: 2
average rate: 1.016
	min: 0.130s max: 4.897s std dev: 1.75380s window: 7
average rate: 1.885
	min: 0.024s max: 4.897s std dev: 1.26481s window: 14
average rate: 2.651
	min: 0.024s max: 4.897s std dev: 1.01414s window: 22
average rate: 3.173
	min: 0.024s max: 4.897s std dev: 0.88482s window: 29
average rate: 3.640
	min: 0.024s max: 4.897s std dev: 0.78400s window: 37
average rate: 3.972
	min: 0.024s max: 4.897s std dev: 0.72002s window: 44
average rate: 4.289
	min: 0.024s max: 4.897s std dev: 0.66254s window: 52
average rate: 4.522
	min: 0.024s max: 4.897s std dev: 0.62245s window: 59
average rate: 4.675
	min: 0.024s max: 4.897s std dev: 0.58851s window: 66
average rate: 4.902
	min: 0.024s max: 4.897s std dev: 0.55614s window: 74
average rate: 5.055
	min: 0.024s max: 4.897s std dev: 0.53162s window: 81
average rate: 5.209
	min: 0.024s max: 4.897s std dev: 0.50722s window: 89
average rate: 5.279
	min: 0.024s max: 4.897s std dev: 0.49126s window: 95
average rate: 5.406
	min: 0.024s max: 4.897s std dev: 0.47185s window: 103
average rate: 5.500
	min: 0.024s max: 4.897s std dev: 0.45660s window: 110
average rate: 5.599
	min: 0.024s max: 4.897s std dev: 0.44088s window: 118
average rate: 5.697
	min: 0.024s max: 4.897s std dev: 0.42670s window: 126
average rate: 5.771
	min: 0.024s max: 4.897s std dev: 0.41536s window: 133
average rate: 5.860
	min: 0.024s max: 4.897s std dev: 0.40348s window: 141
average rate: 5.921
	min: 0.024s max: 4.897s std dev: 0.39384s window: 148
average rate: 5.986
	min: 0.024s max: 4.897s std dev: 0.38362s window: 156
average rate: 6.039
	min: 0.024s max: 4.897s std dev: 0.37530s window: 163
average rate: 6.095
	min: 0.023s max: 4.897s std dev: 0.36663s window: 171
average rate: 6.141
	min: 0.023s max: 4.897s std dev: 0.35955s window: 178
average rate: 6.189
	min: 0.023s max: 4.897s std dev: 0.35192s window: 186
average rate: 6.228
	min: 0.023s max: 4.897s std dev: 0.34567s window: 193
average rate: 6.271
	min: 0.023s max: 4.897s std dev: 0.33873s window: 201
average rate: 6.285
	min: 0.023s max: 4.897s std dev: 0.33305s window: 208
average rate: 6.344
	min: 0.023s max: 4.897s std dev: 0.32712s window: 216
average rate: 6.379
	min: 0.023s max: 4.897s std dev: 0.32123s window: 224
average rate: 6.408
	min: 0.023s max: 4.897s std dev: 0.31633s window: 231
average rate: 6.440
	min: 0.023s max: 4.897s std dev: 0.31100s window: 239
average rate: 6.439
	min: 0.023s max: 4.897s std dev: 0.30723s window: 245
average rate: 6.479
	min: 0.023s max: 4.897s std dev: 0.30239s window: 253
average rate: 6.507
	min: 0.023s max: 4.897s std dev: 0.29773s window: 261
average rate: 6.496
	min: 0.023s max: 4.897s std dev: 0.29460s window: 267
average rate: 6.517
	min: 0.023s max: 4.897s std dev: 0.29104s window: 274
average rate: 6.518
	min: 0.023s max: 4.897s std dev: 0.28752s window: 281
average rate: 6.509
	min: 0.023s max: 4.897s std dev: 0.28462s window: 287
average rate: 6.530
	min: 0.023s max: 4.897s std dev: 0.28122s window: 294
average rate: 6.571
	min: 0.023s max: 4.897s std dev: 0.27726s window: 303
average rate: 6.590
	min: 0.023s max: 4.897s std dev: 0.27426s window: 310
average rate: 6.610
	min: 0.023s max: 4.897s std dev: 0.27094s window: 318
average rate: 6.627
	min: 0.023s max: 4.897s std dev: 0.26815s window: 325
average rate: 6.645
	min: 0.023s max: 4.897s std dev: 0.26491s window: 333
average rate: 6.661
	min: 0.023s max: 4.897s std dev: 0.26217s window: 340
average rate: 6.678
	min: 0.023s max: 4.897s std dev: 0.25915s window: 348
average rate: 6.685
	min: 0.023s max: 4.897s std dev: 0.25658s window: 355
average rate: 6.698
	min: 0.023s max: 4.897s std dev: 0.25409s window: 362
average rate: 6.715
	min: 0.023s max: 4.897s std dev: 0.25133s window: 370
average rate: 6.729
	min: 0.023s max: 4.897s std dev: 0.24899s window: 377
average rate: 6.759
	min: 0.023s max: 4.897s std dev: 0.24615s window: 386
average rate: 6.771
	min: 0.023s max: 4.897s std dev: 0.24409s window: 393
average rate: 6.784
	min: 0.023s max: 4.897s std dev: 0.24164s window: 401
average rate: 6.796
	min: 0.023s max: 4.897s std dev: 0.23956s window: 408
average rate: 6.802
	min: 0.023s max: 4.897s std dev: 0.23753s window: 415
average rate: 6.811
	min: 0.023s max: 4.897s std dev: 0.23528s window: 423
average rate: 6.820
	min: 0.023s max: 4.897s std dev: 0.23336s window: 430
average rate: 6.832
	min: 0.023s max: 4.897s std dev: 0.23122s window: 438
average rate: 6.843
	min: 0.023s max: 4.897s std dev: 0.22940s window: 445
average rate: 6.868
	min: 0.023s max: 4.897s std dev: 0.22719s window: 454
average rate: 6.877
	min: 0.023s max: 4.897s std dev: 0.22557s window: 461
average rate: 6.886
	min: 0.023s max: 4.897s std dev: 0.22376s window: 469
average rate: 6.895
	min: 0.023s max: 4.897s std dev: 0.22211s window: 476
average rate: 6.898
	min: 0.023s max: 4.897s std dev: 0.22049s window: 483
average rate: 6.905
	min: 0.023s max: 4.897s std dev: 0.21869s window: 491
average rate: 6.914
	min: 0.023s max: 4.897s std dev: 0.21715s window: 498

Myzhar, 2020-05-08:
@G-Pereira is `jetson_clocks` running?

G-Pereira, 2020-05-08:
@Myzhar  No idea, how can I tell? I have no running process with that name even after `sudo jetson_clocks`

G-Pereira, 2020-05-08:
I've done some research about power modes and I have tried them all and have run `jetson_clocks` and the problem persists.

Myzhar, 2020-05-08:
Have you changes parameters in `common.yaml` or `zed2.yaml` ?
It's really strange that you get those FPS on a Xavier when we can get more than 15 FPS on a Nano...
does the zed_wrapper node emits any warning?

Another thing that you can try is running `rqt` and visualize the diagnostic information

G-Pereira, 2020-05-08:
> Have you changes parameters in `common.yaml` or `zed2.yaml` ?

Yes, I only enabled object detection. Other than that I am currently turning off tracking and playing with some other parameters (like resolution) to see if the problem goes away. But nothing seems to help, the elaboration time is always around 0.1s

> It's really strange that you get those FPS on a Xavier

Actually is a TX2, but strange anyway yes. I have turned on vehiecle and person detection is that a problem?

> does the zed_wrapper node emits any warning?

Yes:
`[ WARN] [1588950266.471279483]: Elaboration takes longer (0.0957041 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.`

G-Pereira, 2020-05-08:
There is a lot of parameters, can you tell a specific parameter that saves a perceptible amount computing power? Changing resolution doesn't help, which is strange.

Myzhar, 2020-05-08:
It depends on what you activated/changed. Many parameters affect the required computational power.
The warning that you pasted says that something is taking too much time... but I cannot know what it is if you do not paste here the `common.yaml` and `zed2.yaml` files.

G-Pereira, 2020-05-08:
Here it is @Myzhar thank you
[params.zip](https://github.com/stereolabs/zed-ros-wrapper/files/4600293/params.zip)

G-Pereira, 2020-05-11:
@Myzhar I've decided to completely clean my workspace. I have purged my ROS workspace and cloned zed-wrapper and zed-examples again and clean compiled them. I've noticed that when I run:

`roslaunch zed_display_rviz display_zed2.launch`

Without changing any parameters (absolutely no file changed)
I get this while running:

```
[ WARN] [1589192722.651939460]: Elaboration takes longer (0.0350388 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.
[ WARN] [1589192739.089612482]: Elaboration takes longer (0.0338719 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.
[ WARN] [1589192749.164123488]: Elaboration takes longer (0.0403726 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.
[ WARN] [1589192779.218124442]: Elaboration takes longer (0.033705 sec) than requested by the FPS rate (0.033333333 sec). Please consider to lower the 'frame_rate' setting or to reduce the power requirements reducing the resolutions.
```

FYI I am using:
- Jetson TX2 on MAXN power mode
- Jetpack 4.3
- ZED SDK 3.1.2
- ROS Melodic

Should I try Jetpack 4.2 or 4.4 or other version of the ZED SDK?

Myzhar, 2020-05-11:
Can you copy and paste at least one line of the output of the command
`sudo jetson_clocks --show`

Myzhar, 2020-05-11:
However if you run Rviz on the TX2 to show the output of the ZED Wrapper node, it's really hard to reach 30 FPS. 
You are a little over and the warning says exactly that: you cannot reach 30 FPS on the TX2 using that resolution and that framerates.

Are you trying also to display the pointcloud on Rviz?

G-Pereira, 2020-05-11:
Ok, nice to have some idea of the frame rates to expect.

> Are you trying also to display the pointcloud on Rviz?

For the moment, no.

> sudo jetson_clocks --show

I will send you at some point tomorrow thanks

G-Pereira, 2020-05-12:
Here it is:
```
SOC family:tegra186  Machine:quill
Online CPUs: 0-5
CPU Cluster Switching: Disabled
cpu0: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c7=0 
cpu1: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c6=0 c7=0 
cpu2: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c6=0 c7=0 
cpu3: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c7=0 
cpu4: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c7=0 
cpu5: Online=1 Governor=schedutil MinFreq=2035200 MaxFreq=2035200 CurrentFreq=2035200 IdleStates: C1=0 c7=0 
GPU MinFreq=1300500000 MaxFreq=1300500000 CurrentFreq=1300500000
EMC MinFreq=40800000 MaxFreq=1866000000 CurrentFreq=1866000000 FreqOverride=1
Fan: speed=255
NV Power Mode: MAXN
```

G-Pereira, 2020-05-12:
@Myzhar  I don't know if it helps but I get the same issue with the C++ examples (when running the object detection example and tutorial)

Myzhar, 2020-05-12:
@G-Pereira the Jetson Clock is OK, so you are using the maximum available computational power.

The Object Detection modue requires a lot of computational power,  the Jetson TX2 supports it, but you must reduce framerate to 15 and use HD720 (or less) as resolution.
Furthermore I suggest you to not run Rviz on the Jetson, but start it on a remote machine if you want visualize the results of the processing

G-Pereira, 2020-05-13:
@Myzhar Thank you for confirming that the performance I am getting is expected and for the advices and quick support.

Myzhar, 2020-05-13:
You are welcome :)

closed 2020-05-13 by Myzhar

Myzhar, 2023-05-18:
Hi @ashBabu 
please do not reply to a closed issue. Create a new one instead and add a link to the closed issue if required