stereolabs/zed-ros-wrapper #558: [Question] How to adjust the density of the point cloud?

willzoe, 2020-04-30:
I want to use point cloud as input for local costmap. 

Is it possible to downsample a point cloud by setting certain parameters?

Myzhar, 2020-04-30:
Hi @willzoe 
the wrapper does not provide this feature, but you can write a nodelet that subscribes to pointcloud and uses the PCL library to upsample the pointcloud:
http://pointclouds.org/documentation/tutorials/resampling.php
http://docs.pointclouds.org/trunk/classpcl_1_1_bilateral_upsampling.html
http://pointclouds.org/documentation/tutorials/

To write a nodelet you can follow our tutorial:
https://www.stereolabs.com/docs/ros/zed-nodelets/

willzoe, 2020-04-30:
Thank you very much, I will try it. 

> Resample factor for depth data matrices [0.01,1.0] The SDK works with native data sizes, but publishes rescaled matrices (depth map, point cloud, ...)

1. Does this mean that after modifying this parameter, not only the point cloud is downsampled, but the depth map is also downsampled?

I found that rtabmap_ros has a nodelet [point_cloud_xyz](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fpoint_cloud_xyz), which can generate a point cloud from depth or disparity images and downsample it. There are two types of input, one is depth image, and the other is disparity image. 

2. For Zed cameras, which image is better(more accurate results and faster calculations) for input?

Myzhar, 2020-04-30:
1. Yes, using the `depth/depth_downsample_factor` you affect all the data structures involved in the depth estimation (depth map, disparity map, point cloud, Confidence map, ...)

2. You can use both, but depth image does not require conversion from "pixel to depth". However I suggest to not use it since the ZED SDK provides the pointcloud and you add heavy calculations for something that you already have.

willzoe, 2020-04-30:
1. I use the 2.8.5 version of the sdk and its associated zed_wrapper. The parameter `depth / depth_downsample_factor` was not found in the file `common.yaml`.

2. If I don’t subscribe to topics related to depth and point cloud, and only subscribe to the left and right images(`/zed/zed_node/left/image_rect_color` `/zed/zed_node/left/image_rect_color`) and their info(`/zed/zed_node/left/camera_info` `/zed/zed_node/right/camera_info`), will the ZED SDK continue to calculate and publish messages such as depth or point cloud that require a lot of computing resources?

Myzhar, 2020-04-30:
1. You are using a very old version. This parameter is available starting from v3.1
2.  To totally disable the depth module you must set `depth_stabilization` to 0 and `quality` to 0.

willzoe, 2020-04-30:
1. It's really old. I previously encountered some problems when deploying the turtlebot2 related package on melodic, so I changed to kinetic. So I had to use Ubuntu16 and jetpack3.3.

2. Sorry, I did not make it clear just now. In addition to the four topics mentioned above, because nodelet point_cloud_xyz is used, I think I need to use `/zed/zed_node/depth/camera_info` and `/zed/zed_node/depth/camera_info`. In order to reduce the calculation, should I close the topics related to `confidence`, `disparity`, `odom`, `point_cloud` and `pose`?

Myzhar, 2020-04-30:
If a topic is not subscribed the wrapper does not perform computations related to it, to reduce the required computational power simply do not subscribe to topics that you do not use.
If you need `/zed/zed_node/depth/camera_info` you must subscribe to `depth` otherwise it will not be published... however the depth camera info is equal to the left camera camera info

willzoe, 2020-04-30:
Sorry, I wrote it wrong ...
What I need is `/zed/zed_node/depth/camera_info` and `/zed/zed_node/depth/depth_registered`.

Thanks for your answers~

closed 2020-04-30 by willzoe