stereolabs/zed-ros-wrapper #559: [Question] cuda version problem

jehyeonseo, 2020-04-30:
[ INFO] [1588236400.436399525]:  *** Opening ZED...
[ INFO] [1588236400.456698471]: ZED connection -> NVIDIA DRIVER OUT OF DATE
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
in Camera.cpp : void sl::Camera::close() at 158 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in Camera.cpp : void sl::Camera::close() at 202 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
--------------------------------------------------------------------------------------------------------------------------
I just changed the cuda version from 7.5 to 10.0
So is there a problem?
Please help me
Please!

Myzhar, 2020-04-30:
Hi @jehyeonseo 
when you upgrade the CUDA version you must clear the CMake cache before compiling.
Delete the folders `build` and `devel` in you catkin workspace than use the command `catkin_make` to issue a new compilation

jehyeonseo, 2020-04-30:
Thank u.
catkin_ws 폴더에 있는걸 다 삭제하고 다시 깔앗는데도 같은 에러가 나는데 어떡하죠?

Myzhar, 2020-04-30:
Have you also updated the driver of your Nvidia card other than CUDA version?

jehyeonseo, 2020-04-30:
Thanks, it was solved thanks to you! Watch out for COVID 19!

Myzhar, 2020-04-30:
@jehyeonseo you are welcome.
Stay safe 🤞

closed 2020-04-30 by Myzhar

AndreV84, 2020-07-27:
what if it is on 'Jetson? within container? that has cuda pre installed?

Myzhar, 2020-07-27:
@AndreV84 please open a new ticket with more information, never reply to a closed one to not loose tracking.

AndreV84, 2020-07-27:
@Myzhar Thank you for your response!
I raised a bew ticket, but noone responded to there!
https://github.com/stereolabs/zed-docker/issues/24#issue-665892129
https://github.com/stereolabs/zed-docker/issues/22#issuecomment-664062182