stereolabs/zed-ros-wrapper #561: [Question] Estimate depth from rosbag

poornimajd, 2020-05-10:
Hello 
I have a rosbag file recorded using zed camera.
the info is as follows

> path:        /media/smart/My Passport/corridor2_2019-03-26-18-05-02.bag
version:     2.0
duration:    28.4s
start:       May 15 2019 15:39:29.32 (1557914969.32)
end:         May 15 2019 15:39:57.75 (1557914997.75)
size:        146.7 MB
messages:    292519
compression: none [190/190 chunks]
types:       geometry_msgs/PoseStamped   [d3812c3cbc69362b77dc0b19b345f8f5]
             nav_msgs/Odometry           [cd5e73d190d741a2f92e81eda573aca7]
             sensor_msgs/CameraInfo      [c9a58c1b0b154e0e6da7578cb991d214]
             sensor_msgs/CompressedImage [8f7a12909da2c9d3332d540a0977563f]
             sensor_msgs/Imu             [6a62c6daae103f4ff57a132d6f95cec2]
             tf2_msgs/TFMessage          [94810edda583a504dfda3829e70d7eec]
topics:      /tf                                     97596 msgs    : tf2_msgs/TFMessage         
             /tf_static                                  1 msg     : tf2_msgs/TFMessage         
             /zed/imu/data                           90797 msgs    : sensor_msgs/Imu            
             /zed/imu/data_raw                       90795 msgs    : sensor_msgs/Imu            
             /zed/left/camera_info                    3397 msgs    : sensor_msgs/CameraInfo     
             /zed/left/image_rect_color/compressed    3139 msgs    : sensor_msgs/CompressedImage
             /zed/odom                                3397 msgs    : nav_msgs/Odometry          
             /zed/pose                                3397 msgs    : geometry_msgs/PoseStamped

I do not have the depth/depth registered topic.Is it still possible to get the depth map for this video file,using the zed sdk?
I am slightly new to ros,not sure as to how this can be done.
Any suggestion is greatly appreciated.
Thankyou

Myzhar, 2020-05-10:
Hi @poornimajd 
I'm sorry, but the ZED Wrapper uses the ZED SDK to estimate depth and it only accepts RGB data from a connected camera or from a SVO file to extract depth information.

poornimajd, 2020-05-10:
Thank you for the quick reply.

closed 2020-05-10 by poornimajd