stereolabs/zed-ros-wrapper #562: [Question] How to publish bgr8 image

Kimsuhyung, 2020-05-11:
I confirmed that the image subscribed as topic "left/image_rect_color" is in BGRA8 type.

So, I am wondering if I can subscribe the image in BGR8 type.

mZed.retrieveImage(leftZEDMat, sl::VIEW::LEFT, sl::MEM::CPU, mMatResolVideo);

I think it can be solved in this part.

void imageToROSmsg(sensor_msgs::ImagePtr imgMsgPtr, sl::Mat img, std::string frameId, ros::Time t) {

        if(!imgMsgPtr) {            
            return;
        }

        imgMsgPtr->header.stamp = t;
        imgMsgPtr->header.frame_id = frameId;
        imgMsgPtr->height = img.getHeight();
        imgMsgPtr->width = img.getWidth();

        int num = 1; // for endianness detection
        imgMsgPtr->is_bigendian = !(*(char*)&num == 1);

        imgMsgPtr->step = img.getStepBytes();

        size_t size = imgMsgPtr->step * imgMsgPtr->height;
        imgMsgPtr->data.resize(size);

        sl::MAT_TYPE dataType = img.getDataType();

        switch (dataType) {
        case sl::MAT_TYPE::F32_C1: /**< float 1 channel.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::TYPE_32FC1;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::float1>(), size);
            break;

        case sl::MAT_TYPE::F32_C2: /**< float 2 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::TYPE_32FC2;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::float2>(), size);
            break;

        case sl::MAT_TYPE::F32_C3: /**< float 3 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::TYPE_32FC3;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::float3>(), size);
            break;

        case sl::MAT_TYPE::F32_C4: /**< float 4 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::TYPE_32FC4;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::float4>(), size);
            break;

        case sl::MAT_TYPE::U8_C1: /**< unsigned char 1 channel.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::MONO8;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::uchar1>(), size);
            break;

        case sl::MAT_TYPE::U8_C2: /**< unsigned char 2 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::TYPE_8UC2;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::uchar2>(), size);
            break;

        case sl::MAT_TYPE::U8_C3: /**< unsigned char 3 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::BGR8;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::uchar3>(), size);
            break;

        case sl::MAT_TYPE::U8_C4: /**< unsigned char 4 channels.*/
            imgMsgPtr->encoding = sensor_msgs::image_encodings::BGRA8;
            memcpy((char*)(&imgMsgPtr->data[0]), img.getPtr<sl::uchar4>(), size);
            break;
        }
    }

Please let me know if there is a way to solve it.
Thank you.

Myzhar, 2020-05-11:
Hi @Kimsuhyung 
the ZED SDK publishes natively images in the BGRA format to optimize memory.
You can easily convert them in BGR format adding a conversion nodelet,  using for example OpenCV to perform the conversion. 
We will add an example about how to do it soon.
We do not add the conversion directly in the wrapper nodelet to not add elaborations not needed for everyone.

closed 2020-06-23 by Myzhar

lzrrr33, 2020-07-28:
hi @Kimsuhyung Have you solved this problem？ hope to provide some details .thanks.