stereolabs/zed-ros-wrapper #563: [Question] How to enable Neural Depth Sensing for ZED2?

hyylx94, 2020-05-11:
Hi, 

I'm using both ZED mini and ZED2. The ZED SDK I installed is 3.0.1 and I used ROS melodic. 
From the website, I know ZED mini uses stereo depth sensing while ZED2 uses neural stereo depth sensing technology. I would like to know is there any parameter I need to set to switch between these two depth sensing algorithms? Or it will automatically enable neural stereo depth sensing when I set "camera_model:=zed2"?

Thanks in advance!

Kind regards,
Xin

Myzhar, 2020-05-11:
Neural stereo depth sensing is automatically enabled by the SDK using "ULTRA" mode for depth

hyylx94, 2020-05-11:
Thanks for your reply! Then what about ZED mini? If I use ZED mini and set "ULTRA" mode for depth, only normal stereo depth sensing is enabled right?

Myzhar, 2020-05-11:
No, it's available for all the Stereolabs camera models. Only the object detection module is an exclusive of the new ZED2

hyylx94, 2020-05-12:
Thanks! 
One more question about enable "ULTRA" mode, shall I set "3" or "4"?  
In file "param/common.yaml", it seems that I need to set "3":
[image]
In the parameter document, it seems that I need to set "4":
[image]

Myzhar, 2020-05-12:
@hyylx94 `3` is correct. 
Thank you for reporting that the online documentation must be updated :+1:

closed 2020-06-23 by Myzhar