stereolabs/zed-ros-wrapper #564: Incorrect depth values in Zed Depth Viewer

Mizbauddin, 2020-05-19:
Hi,

I am using the Zed Depth Viewer to visualize the point cloud, but the depth values are wrong for various objects in the image. Below is the screenshot. The road is almost 10m far but the values it shows is 3.9m. I was wondering if this is an issue with the camera or the viewer? Thank you. 
[image]

Myzhar, 2020-05-19:
Hi @Mizbauddin 
here we handle issues related to ROS only.
Please write an email to support@stereolabs.com

closed 2020-05-19 by Myzhar

szalontaimartin, 2021-11-17:
Hi @Mizbauddin!

I'd like to inquire about the problem of the distance difference. Did you succeed to solve it? I have the same problem. The distance measured by zed is 5.6 meters, but in fact the distance is 4.8 meters. The greater the distance the greater the difference.

Myzhar, 2021-11-17:
@szalontaimartin this is a closed issue. You should not reply to a closed issue.
Please open a new issue and cite the closed one instead of replying in the same thread.