stereolabs/zed-ros-wrapper #565: [Question] Is it advisable to launch multiple cameras on the same nodelet manager?

tianshiz, 2020-05-19:
Is there any performance issue if I put all 3 of my ZEDs on the same nodelet manager? I'm asking because I'd like my SLAM nodelet to be able to get all 3 sources of image data.

Myzhar, 2020-05-20:
Hi @tianshiz 
I think indeed that it's a good solution to reduce latency getting ZED data for your SLAM processing.

tianshiz, 2020-05-20:
cool, do you know how we can go about doing this? nodelets don't work across different namespaces for me. How would you define 3 cameras within the same namespace?  

normally we'd just make a zed name space for each camera

Myzhar, 2020-05-21:
To use different cameras in the same nodelet manager you must set a different `camera_name` parameter for each of them and you must open them using their `serial_number`.

See [zed_multi_cam.launch](https://github.com/stereolabs/zed-ros-examples/blob/master/examples/zed_multicamera_example/launch/zed_multi_cam.launch)

tianshiz, 2020-05-21:
does that include removing the namespace definition for each camera? see below, if I keep the group ns in place, wouldn't that break the nodelet manager? But, without the namespace, the topic name would be duplicated
```
<node pkg="nodelet" type="nodelet" name="camera_nodelet_manager"  args="manager" output="screen" />
<group ns="zed1_GPU0">
<node pkg="nodelet" type="nodelet" name="node1" args="load zed_nodelets/ZEDWrapperNodelet camera_nodelet_manager" output="screen">
            <arg name="camera_name"                     value="camera1" />
            <arg name="serial_name"                        value="1" />
 </node>
</group>
<group ns="zed2_GPU0">
<node pkg="nodelet" type="nodelet" name="node2" args="load zed_nodelets/ZEDWrapperNodelet camera_nodelet_manager" output="screen">
            <arg name="camera_name"                     value="camera2" />
            <arg name="serial_name"                        value="2" />
 </node>
</group>
```

tianshiz, 2020-05-21:
here's an implementation using a single namespace, but just remapping all the topics manually. Is this the only way to go?
```
<group ns="camera_manager">
<node name="camera_nodelet_manager" pkg="nodelet" type="nodelet" args="manager" output="screen" />
<node name="zed_node" pkg="nodelet" type="nodelet" args="load zed_wrapper/ZEDWrapperNodelet camera_nodelet_manager" >
<rosparam file="$(find mantis_cam)/config/common.yaml" />
<rosparam file="$(find mantis_cam)/config/zed.yaml" />
<param name="svo_file" value="" />
<param name="stream" value="" />
<param name="general/zed_id" value="1" />
<param name="general/gpu_id" value="0" />
<param name="general/serial_number" value="1" />
<param name="general/camera_name" value="1" />
<param name="general/camera_frame" value="zed_center_1" />
<param name="general/left_camera_frame" value="zed_left_camera_1" />
<param name="general/left_camera_optical_frame" value="zed_left_camera_optical_frame_1" />
<param name="general/right_camera_frame" value="zed_right_camera_1" />
<param name="general/right_camera_optical_frame" value="zed_right_camera_optical_frame_1" />
<param name="general/camera_flip" value="false" />
<remap from="rgb/image_rect_color" to="/zed1_GPU0/zed_node/rgb/image_rect_color" />
<remap from="rgb/image_raw_color" to="/zed1_GPU0/zed_node/rgb/image_raw_color" />
<remap from="rgb/camera_info" to="/zed1_GPU0/zed_node/rgb/camera_info" />
<remap from="left/image_rect_color" to="/zed1_GPU0/zed_node/left/image_rect_color" />
<remap from="left/image_raw_color" to="/zed1_GPU0/zed_node/left/image_raw_color" />
<remap from="left/camera_info" to="/zed1_GPU0/zed_node/left/camera_info" />
<remap from="right/image_rect_color" to="/zed1_GPU0/zed_node/right/image_rect_color" />
<remap from="right/image_raw_color" to="/zed1_GPU0/zed_node/right/image_raw_color" />
<remap from="right/camera_info" to="/zed1_GPU0/zed_node/right/camera_info" />
<remap from="depth/depth_registered" to="/zed1_GPU0/zed_node/depth/depth_registered" />
<remap from="point_cloud/cloud_registered" to="/zed1_GPU0/zed_node/point_cloud/cloud_registered" />
<remap from="odom" to="/zed1_GPU0/zed_node/odom" />
</node>
<node name="zed_node" pkg="nodelet" type="nodelet" args="load zed_wrapper/ZEDWrapperNodelet camera_nodelet_manager" >
<rosparam file="$(find mantis_cam)/config/common.yaml" />
<rosparam file="$(find mantis_cam)/config/zed.yaml" />
<param name="svo_file" value="" />
<param name="stream" value="" />
<param name="general/zed_id" value="2" />
<param name="general/gpu_id" value="0" />
<param name="general/serial_number" value="2" />
<param name="general/camera_name" value="2" />
<param name="general/camera_frame" value="zed_center_2" />
<param name="general/left_camera_frame" value="zed_left_camera_2" />
<param name="general/left_camera_optical_frame" value="zed_left_camera_optical_frame_2" />
<param name="general/right_camera_frame" value="zed_right_camera_2" />
<param name="general/right_camera_optical_frame" value="zed_right_camera_optical_frame_2" />
<param name="general/camera_flip" value="false" />
<remap from="rgb/image_rect_color" to="/zed2_GPU0/zed_node/rgb/image_rect_color" />
<remap from="rgb/image_raw_color" to="/zed2_GPU0/zed_node/rgb/image_raw_color" />
<remap from="rgb/camera_info" to="/zed2_GPU0/zed_node/rgb/camera_info" />
<remap from="left/image_rect_color" to="/zed2_GPU0/zed_node/left/image_rect_color" />
<remap from="left/image_raw_color" to="/zed2_GPU0/zed_node/left/image_raw_color" />
<remap from="left/camera_info" to="/zed2_GPU0/zed_node/left/camera_info" />
<remap from="right/image_rect_color" to="/zed2_GPU0/zed_node/right/image_rect_color" />
<remap from="right/image_raw_color" to="/zed2_GPU0/zed_node/right/image_raw_color" />
<remap from="right/camera_info" to="/zed2_GPU0/zed_node/right/camera_info" />
<remap from="depth/depth_registered" to="/zed2_GPU0/zed_node/depth/depth_registered" />
<remap from="point_cloud/cloud_registered" to="/zed2_GPU0/zed_node/point_cloud/cloud_registered" />
<remap from="odom" to="/zed2_GPU0/zed_node/odom" />
</node>
</group>
```

closed 2020-08-24 by Myzhar