stereolabs/zed-ros-wrapper #567: [Question] Difference in Intrinsics parameters:

gaburiero, 2020-05-25:
Hello all,

I have a ZEDm camera and I was trying to obtain the intrinsic parameter of the camera for pose estimation.

- I used the calib.stereolabs website to download these values: https://www.stereolabs.com/developers/calib/?SN=NNNN

- I also used ZED\Explorer program to check these parameters.

- Finally, I used the python API to obtain these values through `get_camera_information().calibration_parameters`

`
cx = zed.get_camera_information().calibration_parameters.left_cam.cx
cy = zed.get_camera_information().calibration_parameters.left_cam.cy
fx = zed.get_camera_information().calibration_parameters.left_cam.fx
fy = zed.get_camera_information().calibration_parameters.left_cam.fy
distortion = zed.get_camera_information().calibration_parameters.left_cam.disto
`

However the values in the two approaches differ a little bit, for example:

[720p left camera from website and ZED_Explorer]
[LEFT_CAM_HD]
fx=699.595
fy=699.595
cx=614.61
cy=346.3235
k1=-0.176666
k2=0.0284369
k3=0
p1=0.000125278
p2=-0.00038134

[720p left camera from python]
fx=668.8412
fy=668.8412
cx=620.3018
cy=346.7469
k1=0.0
k2=0.0
k3=0.0
p1=0.0
p2=0.0

My question is, why the parameters are different when obtained by python code?
Which source is correct? Because I want to be able to obtain these parameters automatically for any new camera, so python would be the best solution for me.

Myzhar, 2020-05-25:
@gaburiero this is not the right place for such a question since this is a repository related to the ROS framework. Next time for generic question like this you can send an email to support@stereolabs.com 

However the reply to your question is simple and can be useful also for other users:
when the ZED SDK opens a camera and starts the frame acquisitions performs a fast recalibration to fix minimal imperfection can be generated by temperature changing, little bumps, vibrations, ...
You can however disable this behaviour to get always the same identical camera parameters, setting to true the parameter [camera_disable_self_calib
](https://www.stereolabs.com/docs/api/structsl_1_1InitParameters.html#affeaa06cfc1d849e311e484ceb8edcc5)

gaburiero, 2020-05-25:
@Myzhar 

I am deeply sorry for this mistake. I really mistaken the repository. I was supposed to ask it in the [zed-python-api ](https://github.com/stereolabs/zed-python-api.git) repo.

Sorry for the inconvenience and thank you for the answer. If you want you can delete the question.

closed 2020-05-25 by gaburiero