stereolabs/zed-ros-wrapper #568: Hi. Can it be used in raspberry pi 4? I want to obtain the skeleton data in system of raspberry pi4 + ros + zed. Looking forward to your reply.

Michael-robo13, 2020-05-29:


Myzhar, 2020-05-29:
Hi @Michael-robo13 
the ZED SDK requires CUDA to run so you cannot use it on a Rpi4 that does not have an Nvidia GPU supporting CUDA.
Furthermore skeleton data processing is high demanding and a Rpi4 has not enough computational power to be reliable for this kind of operations.

closed 2020-05-29 by Myzhar

Michael-robo13, 2020-05-29:
Ok! Thank you so much

maxpolzin, 2021-12-14:
https://github.com/stereolabs/zed-open-capture seems to enable the usage of your ZED camera on a rpi

Myzhar, 2021-12-14:
Hi @maxpolzin 
yes, the ZED Open Capture can run on a Rpi4 without using the ZED SDK. But there is not yet a ROS node exploiting it.

PS this is a closed issue. If you need further help on this topic, please open a new issue, write us an email to support@stereolabs.com, or post the question on the public forum: https://community.stereolabs.com/