stereolabs/zed-ros-wrapper #569: [BUG] substitution args not supported: No Module named 'rospkg'

AeroWRX, 2020-06-01:
**Describe the bug**
Running zed2.launch yields errors seemingly triggered by no module rospkg despite verifying that it is installed to /usr/lib/python3/...

**Device information:**
 - OS: Ubuntu 
 - OS version: 20.04
 - CPU: Intel Xeon E-2176 CPU @ 2.7GHz
 - GPU Nvidia:  Quadro P3200
 - ZED SDK Version: ZED_SDK_Ubuntu18_cuda10.0_v3.1.2.run
 - ROS Wrapper version: Latest  https://github.com/stereolabs/zed-ros-wrapper#build-the-program

**To Reproduce**
Steps to reproduce the behavior:
1. Run roslaunch zed_wrapper zed2.launch
2. Get error

**Expected behavior**
Expected launch file to run

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Running on new install of Ros Noetic with Ubuntu 20.04. Also had to use a different catkin_make command:
catkin_make -DPYTHON_EXECUTABLE=/usr/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/include/python3.7m

Myzhar, 2020-06-01:
Hi @AeroWRX 
I'm sorry, but the ROS Wrapper has not yet been tested with Ubuntu 20.04 and ROS Noetic.
ROS Noetic uses Python3 instead of Python2 and not all the ROS packages work directly on ROS Noetic.

I found this solution for your problem:
https://answers.ros.org/question/245967/importerror-no-module-named-rospkg-python3-solved/

Let me know if it works for you

adimehta03, 2020-06-13:
No @Myzhar , that did not work! Do you think its better to rollback to the previous Ubuntu version?

Myzhar, 2020-06-13:
I think it's better. Furthermore there is no official support for CUDA in Ubuntu 20.04. We are waiting for it to release a version of the ZED SDK for Ubuntu 20.04.

adimehta03, 2020-06-13:
@Myzhar I am a newbie to Ubuntu, is there any way to rollback to the previous version without losing any files?

Myzhar, 2020-06-13:
I think that the only way is formatting. Unless you use a docker image based on Ubuntu 18.04

2273024530, 2020-08-07:
sudo update-alternatives --config python
change your default version of python to python3

closed 2020-08-24 by Myzhar

adamanov, 2020-11-08:
The reason this error occurs is because Python packages in /opt are added to PYTHONPATH environment variable BUT for ones in /usr, path does not exist in PYTHONPATH variable. The python-rospkg gets installed in /usr by default. So you just need following line in .bashrc to fix this:

export PYTHONPATH=$PYTHONPATH:/usr/lib/python3.7/dist-packages

brilliant-ember, 2020-11-09:
This worked for me, because by default python2 is the default so we have to explicitly tell the computer that we want python3 to be default so this fixed it for me.
```
 sudo apt install python-is-python3
```
https://answers.ros.org/question/39657/importerror-no-module-named-rospkg/?answer=363168#post-id-363168

rohitdavas, 2021-02-16:
On ubuntu, the same issue I faced but after `conda deactivate` these issues are resolved.

I think if we have installed `noetic` or any distro as said in guide, the `anaconda` python is creating the problem. 

try `conda deactivate` and it should be fine. 
turn off auto activation of `base` of conda.

awiseturtle, 2021-04-29:
> This worked for me, because by default python2 is the default so we have to explicitly tell the computer that we want python3 to be default so this fixed it for me.
>
> ```
>  sudo apt install [...]

This worked for me too!

LimingGao, 2021-05-05:
I fixed this issue by switching the default python version in my computer from python2.7 to pthon3.5
Switch command:
sudo update-alternatives --config python

cmuell89, 2022-04-14:
I'd be really careful changing the default python as Debian policy is that python = Python 2. update-alternatives is essentially creating a symlink but it could cause problems. Python 2 is deprecated but for anyone running anything but the latest OS, this could be detrimental. adamanov's answer is the correct approach. And make sure your nodes/scripts run Python3 in the shebang.

bernatgaston, 2022-05-05:
None of the above answers worked for me, I had to use
`sudo pip install --target=/opt/ros/noetic/lib/python3/dist-packages rospkg`

djramakrishna, 2022-11-03:
Both @bernatgaston  and @rohitdavas solutions works just fine in my case !