stereolabs/zed-ros-wrapper #572: [Question] mapping + point cloud

FaboNo, 2020-06-19:
I am amazed by the capabilities of ZED2 - we are considering to purchase one for our drone project -  and I would like to ask some questions to clarify some points:

1. I guess you implemented a VIO algorithm to compute the pose of the camera - and did you implement a SLAM or did you accumulate probabilities of computed 3D points in a dedicated 3D data structure (like octomap for instance)?
2. Did you implement loop closure?
3. with RVIZ we can display point cloud, but is it the newly computed point cloud or is it the overall map?
4. Related to 2/ how to access to the whole map - I see from the documentation that there is the rtabmap_ros topic - but how to extract all point clouds from it? The map is needed to plan a path for the drone for instance. Or is it possible to access directly to your 3D data structure?

Thank you very much for answering these questions and congratulation for the amazing work!

Myzhar, 2020-06-23:
Hi @FaboNo 
I'm glad to reply to your questions below:

1. The ZED SDK performs a complete SLAM algorithm with relocation
2. Yes, loop closure is performed using depth and visual information
3. You can display both the real time point cloud and the "fused" pointcloud (see [mapping/fused_cloud](https://www.stereolabs.com/docs/ros/zed-node/#spatial-mapping) topic)
4. You can get the full fused pointcloud as in point 3. But if you need path planning you can use the features of the external RTABmap package. A [complete example is available](https://github.com/stereolabs/zed-ros-examples/tree/master/examples/zed_rtabmap_example)

Do not hesitate to ask for more information

FaboNo, 2020-07-03:
@Myzhar Thank you very much for your message and sorry for my late reply. 
In case we are using RATBmap,  we will use the SLAM from RTAPmap, therefore deactivate the mapping from ZED2, am I correct?
Best regards

Myzhar, 2020-07-03:
Yes, that's correct. 
I suggest you to use the ZED odom topic as source for odometry in RTABmap, like you can find in the example.

Myzhar, 2020-08-24:
@FaboNo can you please update the ticket or close it if you found a solution?

closed 2020-08-24 by FaboNo