stereolabs/zed-ros-wrapper #573: [Question] try to find   the depth value but it returns 'NAN',how to fixed

Zhangeaky, 2020-06-23:


Myzhar, 2020-06-23:
Hi @Zhangeaky ,
if it return "NaN" it means that no depth info are available for that pixel of the depth map.
This is expected. Not all the points of a RGB image can have a depth due to occlusions (shadows), environment lights, reflections, uniform textures and many other factors.

Zhangeaky, 2020-06-23:
Few days ago  I  learned from google  that  camera ZED can estimate the depth value(if a pixel point returns 'NAN') by modify a parameter  in launch file of the ros package,and I want to know , can camera ZED2  do that ? because I don't find that paramter in zed2's  launch file. Thank you all the way!

Myzhar, 2020-06-23:
That parameter is used in VR but it's not available anymore for robotics applications because it introduces depth values that do not exist in the real world. It is SENSING_MODE::FILL
You can read more here:
https://www.stereolabs.com/docs/api/group__Depth__group.html#ga391147e2eab8e101a7ff3a06cbed22da

closed 2020-07-01 by Zhangeaky