stereolabs/zed-ros-wrapper #575: [Question] ZED 2 camera disconnects many times

cnpcshangbo, 2020-06-23:
Hi,
I just replace ZED with ZED 2 with the same Jetson TX2 system. However, I found that the camera disconnects too often. 
[image]
Are there something I can do to improve?
Thanks,
Bo

Myzhar, 2020-06-23:
Hi @cnpcshangbo 
can you provide me with more information about the versions of SDK, ZED Wrapper and Nvidia Jetpack that you are using?

cnpcshangbo, 2020-06-23:
Hi @Myzhar, SDK version is v3.1.2 Do I need to upgrade to the latest SDK? Thanks,
Bo

cnpcshangbo, 2020-06-23:
I haven't use ZED Wrapper. I am using the positional tracking sample from ZED.

cnpcshangbo, 2020-06-23:
@Myzhar I have tried the latest version of ZED SDK. It often takes a lot of trials to connect to the camera and very easy to get disconnected. Hope to get your suggestions. Thanks.

Myzhar, 2020-06-23:
Please write an email to support@stereolabs.com to open a ticket in order to better handle your issue.
I close this since it's not related to ROS

closed 2020-06-23 by Myzhar