stereolabs/zed-ros-wrapper #576: [Question] Do I have to undistort the images obtained from Zed camera?

ShrutheeshIR, 2020-06-27:
Hello! I am fairly new to this sphere, so apologies in advance if the question is redundant.

I have taken a video sequence using the Zed camera. 
```

fx=349.9
fy=349.9
cx=335.3375
cy=199.2915
k1=-0.177001
k2=0.0291024
k3=0
p1=-0.00192826
p2=0.000284981

```
These are my camera intrinsics parameters.
In another comment, I read that the zed camera gives out rectified images. So what do rectified images mean?

Before processing, do I have to undistort the points using k1,k2,p1,p2
Or can I ignore the lens distortion parameters and continue?
Note: I am trying to implement a custom SLAM algorithm and hence these parameters matter a lot.

Myzhar, 2020-06-27:
Hi @ShrutheeshIR 
it depends on the topics that you are subscribing.
Topics with `_raw` must be rectified, topics without `_raw` are ready to be used:
https://www.stereolabs.com/docs/ros/zed-node/#published-topics

ShrutheeshIR, 2020-06-27:
Thanks @Myzhar I am subscribing to ```rgb/image_rect_color```, and hence the rectified images.
My query is: 
Can I just ignore the ```k1```,```k2```,```p1```,```p2``` values and use the ```fx```, ```fy```, ```cx```, ```cy``` directly, or should I make any changes?

Myzhar, 2020-06-27:
fx, fy, cx and cy are required to extract 3D info.
The distortion parameters are null for rectified images

ShrutheeshIR, 2020-06-27:
@Myzhar Thanks!
Just to confirm once again, the fx, fy, cx and cy values remain the same for both rectified and unrectified images right?

Myzhar, 2020-06-27:
Exactly 👍

ShrutheeshIR, 2020-06-27:
Thanks a ton!

closed 2020-06-27 by ShrutheeshIR