stereolabs/zed-ros-wrapper #577: Duplicate imu timestamps

Abdob, 2020-06-28:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**
 - OS: [e.g. Ubuntu]
 - OS version: [e.g. 18.04]
 - CPU: [e.g. i7, Jeston Nano, Jetson TX2]
 - GPU Nvidia:  [e.g. GTX1080, RTX1070]
 - ZED SDK Version: [e.g. v3.1.2, v2.8.5]
 - ROS Wrapper version: [e.g. v2.8.5, latest master]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Abdob, 2020-06-28:
I'm running the zed cam and launching the rostopics. I am not using the time sync mode between imu and camera. I record the imu data into a bag. Some imu timestamps occur twice

Myzhar, 2020-06-28:
Hi @Abdob 
Can you provide all the information required by the issue?
We will investigate the issue

Abdob, 2020-06-28:
Hi @Myzhar  I have to mention it is the zed2 camera, although it also happens on the jetson nx i brought forth the information on my desktop system.

Thanks,

JarvisMark1, 2020-07-25:
I had the same problem with ZED mini。

tftangming, 2020-08-13:
I meet the same problem and I want to know whether you have solved the problem? Thanks you very much

Myzhar, 2020-08-31:
Fixed. Will be merged soon into master

closed 2020-11-05 by Myzhar