stereolabs/zed-ros-wrapper #580: [Question] How to use ZED SDK without CUDA?

violet17, 2020-07-03:
Hi,
 I found that the SDK could be installed without CUDA,  and images could be obatined from the camera. Could the depth map be computed by CPU without CUDA?

Thanks.

Myzhar, 2020-07-03:
Hi @violet17 
you can install the SDK without CUDA, but you cannot use it to acquire images and extract depth because the underlying processing in entirely based on CUDA.
The only way to use a ZED without CUDA is exploting the [ZED Open Capture library](https://github.com/stereolabs/zed-open-capture), but a ROS Wrapper has not been yet developed around it. 
Furthermore extracting the depth map on a CPU, even if it can be performed, isn't useful for a real time robotic application unless you use very low image resolutions.
ROS provides node to extract the depth map from a stereo couple: 
http://wiki.ros.org/stereo_image_proc

violet17, 2020-07-03:
Hi @Myzhar , 
Thank you very much for your early response.
Is there any way to compute the depth map in the ZED Open Capture library? 
I don't think I need to compute the depth map of whole image. In this case, i7 CPU may be enough for real time computation. 
It is so sad that ZED camera cannot directly support the other platform, which prevents some people to use it.

Myzhar, 2020-07-03:
You can use [OpenCV to process the stereo couple](https://docs.opencv.org/4.3.0/dd/d53/tutorial_py_depthmap.html).
The [rectify example](https://github.com/stereolabs/zed-open-capture/blob/master/examples/zed_oc_rectify_example.cpp)  is a good starting point.

Please open a new ticket on the "ZED Open Capture" repository if you need more info about this part.

violet17, 2020-07-03:
@Myzhar 
Thanks for the response.

closed 2020-07-03 by violet17

armando-fandango, 2026-11-02:
can the processing be done on NPU instead of GPU ? does the SDK support NPU ?