stereolabs/zed-ros-wrapper #581: heavy swings in frame arrivals (timestamp differences)

Abdob, 2020-07-06:
When setting the frame grab rate to 30fps I'm seeing timestamp differences between 18 ms and 47ms. 

Ideally we'd like to garauntee 33 +- 3 ms between every two frames. Is there a way to mitigate this with the zed-ros-wrapper or the zed-open-capture?

Myzhar, 2020-07-06:
Hi @Abdob 
Can you give me more info about your system?
Usually the timestamps are very precise.

Abdob, 2020-07-06:
Hi @Myzhar 

It is running on jetson nano jetpack 4.3. Zed2 cam latest sdk and latest ros wrapper. The images were grabbed using rosbag record /zed2/zed_node/image_left/image_raw_gray same for right and a couple other topics included.

Myzhar, 2020-07-06:
Ok, the problem is rosbag file saving.
It takes too many resources on a Jetson Nano and the grabbing threads are slowed down. That's why timestamps are not constant

Myzhar, 2020-07-06:
Have you enabled the maximum power mode on Jetson Nano?

Abdob, 2020-07-06:
I will double check shortly, does the broadcasting of many topics consume resources? If so it would make sense to limit the number of topics to just the ones which are needed

Myzhar, 2020-07-06:
A topic is broadcasted only a node subscribes it.

Abdob, 2020-07-06:
yes the max power is enabled. 
The usb bandwidth utilization is about the same with or without a subscribed node:
Bus ID 2 (USB bus number 2)	To device	From device
  Device ID 6 :			53.87 KiB/s	27263.68 KiB/s

Myzhar, 2020-07-10:
@Abdob you are using a Jetson Nano at 30 FPS, can I know the resolution that you are using?
The Nano can handle 30 FPS or more only at low resolutions, but it mostly depends on the modules you activated.

Myzhar, 2020-08-24:
@Abdob can you please update the ticket or close it if you found a solution?

Abdob, 2020-08-24:
@Myzhar I was able to improve this with maximizing the jetson clock frequencies and disabling usb auto suspend on the jetson nano. This was even more improved on the jetson nx, however it was still a problem for USB cameras especially with dropped frames. I had a colleague do this experiment with a realsense camera and the results looked better. Ultimately I ended up using a camera with MIPI interface and a separate IMU for my purpose.

closed 2020-08-24 by Abdob