stereolabs/zed-ros-wrapper #582: [Question] Position of detected bounding box

jorgemia, 2020-07-06:
Sorry I'm waiting for my camera to arrive and could probably figure this out when I get the camera, but trying to adapt my code so it works asap. 

The objects_stamped msg provides a **position** of the identified object as a Point32. How are the axes defined? Is depth/distance from the person to the camera on the **_z_** axis of this message?

And also, is this depth calculated by simply sampling the depth at the centre of the bounding box (retrieving a single value from the depth image)? If, for example, the camera detected a person and only the legs are in the field of view, the centroid of the bounding box might be between both legs at times. I know this wouldn't happen often but could this give incorrect depth values at times?

Myzhar, 2020-07-06:
Hi @jorgemia 

the axes are defined according to the ROS standard: X Forward, Y LEFT, Z UP, so the distance of an object from the camera is on the X axis.
The position of the object is the centroid of the positions off all the 3D points that compose the object itself.
In case of partial object detection the centroid is calculated according to the visible data, if the tracking is active and the partial object matches an previously seen object, the centroid position is "smoothed".

jorgemia, 2020-07-06:
Great thanks! So I imagine the X axis value will be the mean of all the depths of the points of the object? And smoothing will occur when there are occlusions etc

When you say all the 3D points that compose the object itself...does that refer to all the points inside the bounding box or all the points inside eg. the mask of a person?

jorgemia, 2020-07-06:
Ok! Thank you

closed 2020-07-06 by jorgemia