stereolabs/zed-ros-wrapper #583: Can I get tf updates without subscribe topic \pose?

cnpcshangbo, 2020-07-07:
Hi,
I notices that we must subscribe to the pose topic to get the related tf updates. Is there a specific reason for designing like this?
https://github.com/stereolabs/zed-ros-wrapper/blob/636f019fa0334cd22a2cb0dfadda87b37185df70/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L3501

However, we just need the tf information. Do you recommend us to make a node that subscribes pose topic and do nothing or you have a better way to make tf updates without subscribing pose topic?
Thanks,
Bo

Myzhar, 2020-07-08:
Hi @cnpcshangbo 
actually the comment you cited is not fully true (it was so in the past), the conditions that cause the "positional tracking" to start, and so the TF updating are many and you can find them in those part of code:
https://github.com/stereolabs/zed-ros-wrapper/blob/636f019fa0334cd22a2cb0dfadda87b37185df70/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L3019-L3020

This is however an old behavior and I agree with you that `odom` and `pose` tf publishing should be activated automatically setting to `true` a node param or using a service, like it works for other node sub-modules (e.g. object detection or 3D mapping).

We will add this new feature in future releases of the node.

Myzhar, 2020-08-31:
Available with https://github.com/stereolabs/zed-ros-wrapper/commit/08d1ef615acf05a200d7920808fbbd82b848a0c2
Will be merged soon into master

cnpcshangbo, 2020-09-03:
Sounds great. Thanks.

closed 2020-09-03 by Myzhar