stereolabs/zed-ros-wrapper #584: [Question] Should I define the position of my camera in the launch file or urdf file?

jorgemia, 2020-07-09:
The zed2.launch, zed_camera.launch and urdf file all have the arguments like `_cam_pos_x, cam_pos_y`... available to edit.

Where is the right place to define the link between the camera and base link (i.e. where should I put my numbers)?

I understand the flow of information when launching zed2.launch is:

zed2.launch -> zed_camera.launch -> urdf

So if I put the measurements into zed2.launch it should work? But that might not be the most obvious place to have it so can I delete the arguments from the launch files and just define it in the urdf file? It should just be a static link for a simple 4-wheel skid steer robot.

jorgemia, 2020-07-09:
I've modified the zed_camera.launch file to the following below. I believe I no longer need the zed2.launch file? As the only thing that it does is set the `camera_model` to zed2? 

I deleted the `publish_urdf` statement and all related arguments from the launch file as the robot state publisher will be started from a different launch file and I have defined all the translations inside the actual urdf file of the camera. Is this approach appropriate?

```
<launch>
    <!-- Camera Model and Name -->
    <arg name="camera_name"           default="zed" /> <!-- The name you want -->
    <arg name="camera_model"          default="zed2" /> <!-- 'zed' or 'zedm' or 'zed2' -->
    <arg name="node_name"             default="zed_node" />

    <!-- Load SVO file -->
    <arg name="svo_file"              default="" /><!-- <arg name="svo_file" default="path/to/svo/file.svo"> -->
    <!-- Remote stream -->
    <arg name="stream"                default="" /> <!-- <arg name="stream" default="<ip_address>:<port>"> -->

    <!-- Base frame -->
    <arg name="base_frame"            default="base_link" />

    <arg name="camera_id"             default="0" />
    <arg name="gpu_id"                default="-1" />

    <node name="$(arg node_name)" pkg="zed_wrapper" type="zed_wrapper_node" output="screen" required="true"><!-- launch-prefix="valgrind" -->
        <rosparam file="$(find zed_wrapper)/params/common.yaml" command="load" />
        <rosparam file="$(find zed_wrapper)/params/$(arg camera_model).yaml" command="load" />

        <!-- Camera name -->
        <param name="general/camera_name"               value="$(arg camera_name)" />

        <!-- Base frame -->
        <param name="general/base_frame"                value="$(arg base_frame)" />

        <!-- SVO file path -->
        <param name="svo_file"                          value="$(arg svo_file)" />

        <!-- Remote stream -->
        <param name="stream"                            value="$(arg stream)" />

        <!-- Camera ID -->
        <param name="general/zed_id"                    value="$(arg camera_id)" />

        <!-- GPU ID -->
        <param name="general/gpu_id"                    value="$(arg gpu_id)" />
    </node>
</launch>
```

Lastly, I've been reading #447 . If I understood correctly, the ZED automatically transforms the odom/pose data to make it relative to the base_link instead of the camera centre? That means I can use the odom data in another node as if it was at the centre of the robot (don't need to apply any transforms)?

Myzhar, 2020-07-09:
Hi @jorgemia 
The launch files that you find in the repository are designed to be as general al possible to be usable in each kind of configuration. You are free to modify them in order to get the best configuration that is good for your robot configuration.
Be aware that there are a few variables that must be set in order to get a valid configuration of the URDF that is generated by a XACRO script.
Indeed the camera position and orientation is set in the launch file and propagated in the URDF using XACRO variables.

The "zed_no_tf.launch" file simply starts the node disabling odom and pose tf publication. It is important if you use another source of odometry and no sensor fusion nodes.

jorgemia, 2020-07-09:
@Myzhar what variables are you referring to? I think I got the urdf working with what I said as I can visualise the camera and my robot in rviz. I simply removed the base_link definition (to avoid defining it twice) in the `zed_descr.urdf.xacro` and imported the file in my main robot urdf file using the `include` command.

Does the ZED2 accept wheel odometry inputs? Or should I use an external package (robot_localization) for that? Should I disable tf publication then?

Myzhar, 2020-07-09:
The variables are declared here:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/urdf/zed_descr.urdf.xacro#L29-L37

For external odometry sources you need a fusion node. EKF or UKF in robot_localization are perfect

jorgemia, 2020-07-09:
Ok yeah I've got those variables! Thanks!

So if I fuse the odom from the camera with the odom from the wheels using the EKF, will I have any problems with how the transforms are set up right now? Is the odom provided by the camera centred in the robot or the camera? (Sorry for all the questions that might be obvious)

Myzhar, 2020-07-09:
The odometry is internally transformed to `base_link`.
The node gets the relative tfs and automatically performs all the calculations

jorgemia, 2020-07-09:
Ok thank you!

closed 2020-07-09 by jorgemia