stereolabs/zed-ros-wrapper #585: [BUG]

CMartinETH, 2020-07-10:
**Describe the bug**
According to the description on the stereolabs homepage (https://www.stereolabs.com/docs/ros/zed-node/) it is possible to change the resolution of the camera with the general -> resolution parameter. I added this parameter and changed it to "VGA", but I keep getting the error: 
[ INFO] [1594370263.065620014]:  * Camera Resolution		-> Unknown
[ WARN] [1594370263.065828595]: Invalid resolution. Set to HD720 @ 30 FPS
I also tried the other resolution settings (e.g. HD1080 etc.) and it seems that I cannot change this. All other settings (framerate etc) I tried, do work. 

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: AMD Ryzen 9 3900X
 - GPU Nvidia:  RTX2080 8GB
 - ZED SDK Version: 3.2.0 (newest)
 - ROS Wrapper version: latest master

**Expected behavior**
The resolution should change accordingly. 

**Screenshots**
[image]

closed 2020-07-10 by CMartinETH

Myzhar, 2020-07-10:
Hi @CMartinETH 
as illustrated in the comments the resolution parameter is an "integer value" not a "string":
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L26
so to set VGA resolution you must use
`resolution:     3`

A note, also `frame_rate` is an integer parameter, to set 15 FPS you must use
`frame_rate:   15`

CMartinETH, 2020-07-10:
The integers work, but not when written in quotes '' (at least in my case). Writing 3 (for VGA) without quotes works fine.

Myzhar, 2020-07-10:
Yes, if you use the quotes the value is identified as a string

CMartinETH, 2020-07-10:
Understood, but why does it work with quotes for the frame_rate then and not the resolution? Pure luck? 
Thank you anyway for the quick help!

Myzhar, 2020-07-10:
Because the default value hard coded for the frame_rate is 15 FPS, the not valid value in the YAML file is replaced with 15 FPS.

**Note**: according to the names of your parameters you are using an old version of the wrapper.
If you update to a newer version please take care of the name changing:
`frame_rate` -> `grab_frame_rate`
The new name has been introduced to distinguish between "grab" rate and "publish" rate. 
See the new `pub_frame_rate` dynamic parameter: http://wiki.ros.org/zed-ros-wrapper#Dynamic_Parameters

CMartinETH, 2020-07-10:
@Myzhar Actually the wrapper is the newest version available. But I took the "frame_rate" from your homepage where the options are explained (here: https://www.stereolabs.com/docs/ros/zed-node/). That means that should be updated for future reference.

Myzhar, 2020-07-10:
Sorry for the mismatch, I fix it as soon as possible, thank you for reporting