stereolabs/zed-ros-wrapper #586: [BUG] RVIZ ERROR  PluginlibFactory: The plugin for class 'rviz_imu_plugin/Imu' failed to load

FaboNo, 2020-07-10:
**Describe the bug**
A clear and concise description of what the bug is.
 
When I enter `roslaunch zed_display_rviz display_zed2.launch`
I have systematically the same error from rviz:

`[ERROR] [1594381653.770729745]: PluginlibFactory: The plugin for class 'rviz_imu_plugin/Imu' failed to load.  Error: According to the loaded plugin descriptions the class rviz_imu_plugin/Imu with base class type rviz::Display does not exist. Declared types are  rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu

Do you hae any idea from where it comes from?`
**Device information:**
 - OS: UBUNTU
 - OS version:18.04
 - CPU: Jetson Nano
 - ZED SDK Version: I do not not know how to find the version of the sdk - just downloaded it last week
 - ROS Wrapper version:  latest master

Regards

Myzhar, 2020-07-10:
Have you installed the `rviz_imu_plugin`?
`$ sudo apt install ros-melodic-rviz-imu-plugin`

It should be automatically installed when you exec the command
`$ rosdep install --from-paths src --ignore-src -r -y`
while installing the zed-ros-examples package

FaboNo, 2020-07-10:
Yes got it!
Thank you

closed 2020-07-10 by FaboNo