stereolabs/zed-ros-wrapper #587: [Question] Does ZED 2 starts assuming it faces East?

cnpcshangbo, 2020-07-11:
I noticed that ZED 2 thinks it faces East when it starts. Why the magnetic information is not used to correct its heading to the exact direction?
Thanks,
Bo

Myzhar, 2020-07-11:
Hi @cnpcshangbo 
The node starts with value 0 for yaw, so it's facing east.
The magnetometer is not used to set an absolute yaw value for two main reasons:
1) not everyone needs to start with an absolute yaw value
2) the magnetometer readings highly depends on where the camera is placed on a robot and on what kind of magnetic fields noises are present in the position it is placed

The reason 2 is the most important of the two, to use the magnetometer it is required that hard and soft iron calibrations are performed after that the camera is placed on the robot and only after that the readings are corrected with the calibration values you can use them as an absolute yaw measure.
This is why we cannot provide such kind of factory calibration and this is left to users.

cnpcshangbo, 2020-07-11:
Thanks for your clarification.

closed 2020-07-27 by Myzhar