stereolabs/zed-ros-wrapper #588: [BUG] Warnings with catkin_make on Jetpack 4.4. Production Release

jorgemia, 2020-07-12:
**Device information:**
 - OS: Ubuntu
 - OS version: 18.04 (Jetpack 4.4 Production Release - L4T R32.4.3)
 - CPU: Xavier NX
 - GPU Nvidia:  Xavier NX
 - ZED SDK Version: 3.2
 - ROS Wrapper version: Latest

**Describe the bug**
I just got a new Xavier NX and updated it to L4T R32.4.3. I was able to install the ZED SDK fine and the ros wrapper also built successfully after running catkin_make. I, however, got some warnings when running catkin_make I wanted to make you aware of. I don't remember these happening when I built on another Xavier NX running L4T R32.4.2 so don't know if the new update might have caused some issues?

```
[ 98%] Building CXX object zed-ros-wrapper/zed_nodelets/CMakeFiles/ZEDNodelets.dir/src/rgbd_sensors_sync_nodelet/src/rgbd_sensor_sync.cpp.o
In file included from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/include/sl_tools.h:26:0,
                 from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:24,
                 from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp: In constructor ‘sl::RuntimeParameters::RuntimeParameters(sl::SENSING_MODE, bool, int, int, sl::REFERENCE_FRAME)’:
/usr/local/zed/include/sl/Camera.hpp:5646:63: warning: ‘sl::RuntimeParameters::textureness_confidence_threshold’ is deprecated: use texture_confidence_threshold instead [-Wdeprecated-declarations]
         , measure3D_reference_frame(measure3D_reference_frame_) {
                                                               ^
/usr/local/zed/include/sl/Camera.hpp:5623:48: note: declared here
         int textureness_confidence_threshold = 100;
                                                ^~~
/usr/local/zed/include/sl/Camera.hpp:5646:63: warning: ‘sl::RuntimeParameters::textureness_confidence_threshold’ is deprecated: use texture_confidence_threshold instead [-Wdeprecated-declarations]
         , measure3D_reference_frame(measure3D_reference_frame_) {
                                                               ^
/usr/local/zed/include/sl/Camera.hpp:5623:48: note: declared here
         int textureness_confidence_threshold = 100;
                                                ^~~
/usr/local/zed/include/sl/Camera.hpp: In destructor ‘sl::CameraInformation::~CameraInformation()’:
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::calibration_parameters’ is deprecated: Use camera_configuration.calibration_parameters instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3193:136: note: declared here
 .calibration_parameters instead")/*@endcond*/ CalibrationParameters calibration_parameters; /**< \deprecated see CameraConfiguration::calibration_parameters, Intrinsic and Extrinsic stereo parameters for rectified/undistorded images (default).*/
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::calibration_parameters’ is deprecated: Use camera_configuration.calibration_parameters instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3193:136: note: declared here
 .calibration_parameters instead")/*@endcond*/ CalibrationParameters calibration_parameters; /**< \deprecated see CameraConfiguration::calibration_parameters, Intrinsic and Extrinsic stereo parameters for rectified/undistorded images (default).*/
                                                                     ^~~~~~~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::calibration_parameters_raw’ is deprecated: Use camera_configuration.calibration_parameters_raw instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3194:140: note: declared here
 ibration_parameters_raw instead")/*@endcond*/ CalibrationParameters calibration_parameters_raw; /**< \deprecated see CameraConfiguration::calibration_parameters_raw, Intrinsic and Extrinsic stereo parameters for original images (unrectified/distorded).*/
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::calibration_parameters_raw’ is deprecated: Use camera_configuration.calibration_parameters_raw instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3194:140: note: declared here
 ibration_parameters_raw instead")/*@endcond*/ CalibrationParameters calibration_parameters_raw; /**< \deprecated see CameraConfiguration::calibration_parameters_raw, Intrinsic and Extrinsic stereo parameters for original images (unrectified/distorded).*/
                                                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::camera_imu_transform’ is deprecated: Use sensors_configuration.camera_imu_transform instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3195:127: note: declared here
 figuration.camera_imu_transform instead")/*@endcond*/ sl::Transform camera_imu_transform; /**< \deprecated see SensorsConfiguration::camera_imu_transform, IMU to Left camera transform matrix, that contains rotation and translation between IMU frame and camera frame.*/
                                                                     ^~~~~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3186:12: warning: ‘sl::CameraInformation::camera_imu_transform’ is deprecated: Use sensors_configuration.camera_imu_transform instead [-Wdeprecated-declarations]
     struct CameraInformation {
            ^~~~~~~~~~~~~~~~~
/usr/local/zed/include/sl/Camera.hpp:3195:127: note: declared here
 figuration.camera_imu_transform instead")/*@endcond*/ sl::Transform camera_imu_transform; /**< \deprecated see SensorsConfiguration::camera_imu_transform, IMU to Left camera transform matrix, that contains rotation and translation between IMU frame and camera frame.*/
                                                                     ^~~~~~~~~~~~~~~~~~~~
/home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp: In member function ‘virtual void zed_nodelets::ZEDWrapperNodelet::onInit()’:
/home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:271:50: note: synthesized method ‘sl::CameraInformation::~CameraInformation()’ first required here 
     mZedRealCamModel = mZed.getCameraInformation().camera_model;
                                                  ^
[... 193 lines cut ...]
In file included from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/include/sl_tools.h:26:0,
                 from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/include/zed_wrapper_nodelet.hpp:24,
                 from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:21:
/usr/local/zed/include/sl/Camera.hpp:3115:112: note: declared here
 Use stereo_transform.getRotationMatrix instead")/*@endcond*/ float3 R; /**< \deprecated see stereo_transform, Rotation on its own (using Rodrigues' transformation) of the right sensor. The left is considered as the reference. Defined as 'tilt', 'convergence' and 'roll'. Using a \ref Rotation, you can use \ref Rotation::setRotationVector(R) to convert into other representations.*/
                                                                     ^
/home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp:1980:73: warning: ‘sl::CalibrationParameters::R’ is deprecated: Use stereo_transform.getRotationMatrix instead [-Wdeprecated-declarations]
             std::vector<float> R_ = sl_tools::convertRodrigues(zedParam.R);
                                                                         ^
In file included from /home/unmnd/catkin_ws/src/zed-ros-wrapper/zed_nodelets/src/tools/include/sl_tools.h:26:0,
[... cut ...]
/usr/local/zed/include/sl/Camera.hpp:3115:112: note: declared here
 Use stereo_transform.getRotationMatrix instead")/*@endcond*/ float3 R; /**< \deprecated see stereo_transform, Rotation on its own (using Rodrigues' transformation) of the right sensor. The left is considered as the reference. Defined as 'tilt', 'convergence' and 'roll'. Using a \ref Rotation, you can use \ref Rotation::setRotationVector(R) to convert into other representations.*/
                                                                 
```

Myzhar, 2020-07-12:
Hi @jorgemia 
They are only warning that will disappear with the future releases of the ZED SDK. You can ignore them 👍

jorgemia, 2020-07-12:
Great, thanks

closed 2020-07-12 by jorgemia

Myzhar, 2020-07-15:
Fixed with https://github.com/stereolabs/zed-ros-wrapper/pull/590