stereolabs/zed-ros-wrapper #589: [BUG]something wrong between the zedmini and ros?

peter2527, 2020-07-13:
**Describe the bug**
A clear and concise description of what the bug is.

**Device information:**zedmini
 - OS: [e.g. Ubuntu]ubuntu
 - OS version: [e.g. 18.04]18.04
 - CPU: [e.g. i7, Jeston Nano, Jetson TX2]i7-1075H
 - GPU Nvidia:  [e.g. GTX1080, RTX1070]GTX1660TI
 - ZED SDK Version: [e.g. v3.1.2, v2.8.5]3.20
 - ROS Wrapper version: [e.g. v2.8.5, latest master]laster master

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'run vins-fusion with zedmini
2. Click on '....'when i launch zedmini
3. Scroll down to '....'include zed/rviz/config file
4. See error:time t:-nan -nan -nan q:nan nan nan

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

peter2527, 2020-07-13:
roslaunch zed_wrapper zedm.launch

... logging to /home/peter/.ros/log/f60c0660-c521-11ea-9b8d-ccf9e45b891b/roslaunch-peter-Y7000-2020H-6050.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://peter-Y7000-2020H:45141/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6
 * /zedm/zed_node/auto_exposure_gain: True
 * /zedm/zed_node/auto_whitebalance: True
 * /zedm/zed_node/brightness: 4
 * /zedm/zed_node/contrast: 4
 * /zedm/zed_node/depth/depth_downsample_factor: 1.0
 * /zedm/zed_node/depth/depth_stabilization: 1
 * /zedm/zed_node/depth/max_depth: 10.0
 * /zedm/zed_node/depth/min_depth: 0.35
 * /zedm/zed_node/depth/openni_depth_mode: 0
 * /zedm/zed_node/depth/quality: 1
 * /zedm/zed_node/depth/sensing_mode: 0
 * /zedm/zed_node/depth_confidence: 50
 * /zedm/zed_node/depth_texture_conf: 100
 * /zedm/zed_node/exposure: 100
 * /zedm/zed_node/gain: 100
 * /zedm/zed_node/gamma: 8
 * /zedm/zed_node/general/base_frame: base_link
 * /zedm/zed_node/general/camera_flip: False
 * /zedm/zed_node/general/camera_model: zedm
 * /zedm/zed_node/general/camera_name: zedm
 * /zedm/zed_node/general/gpu_id: -1
 * /zedm/zed_node/general/grab_frame_rate: 15
 * /zedm/zed_node/general/resolution: 2
 * /zedm/zed_node/general/self_calib: True
 * /zedm/zed_node/general/serial_number: 0
 * /zedm/zed_node/general/svo_compression: 2
 * /zedm/zed_node/general/verbose: False
 * /zedm/zed_node/general/zed_id: 0
 * /zedm/zed_node/hue: 0
 * /zedm/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zedm/zed_node/mapping/mapping_enabled: False
 * /zedm/zed_node/mapping/max_mapping_range: -1
 * /zedm/zed_node/mapping/resolution: 0.05
 * /zedm/zed_node/point_cloud_freq: 15.0
 * /zedm/zed_node/pos_tracking/area_memory: True
 * /zedm/zed_node/pos_tracking/area_memory_db_path: 
 * /zedm/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zedm/zed_node/pos_tracking/floor_alignment: False
 * /zedm/zed_node/pos_tracking/imu_fusion: True
 * /zedm/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zedm/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zedm/zed_node/pos_tracking/map_frame: map
 * /zedm/zed_node/pos_tracking/odometry_frame: odom
 * /zedm/zed_node/pos_tracking/path_max_count: -1
 * /zedm/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zedm/zed_node/pos_tracking/publish_map_tf: True
 * /zedm/zed_node/pos_tracking/publish_tf: True
 * /zedm/zed_node/pos_tracking/two_d_mode: False
 * /zedm/zed_node/pub_frame_rate: 15.0
 * /zedm/zed_node/saturation: 4
 * /zedm/zed_node/sensors/sensors_timestamp_sync: False
 * /zedm/zed_node/sharpness: 4
 * /zedm/zed_node/stream: 
 * /zedm/zed_node/svo_file: 
 * /zedm/zed_node/video/extrinsic_in_camera_frame: True
[... 130 lines cut ...]
[ INFO] [1594656028.312720503]: Advertised on topic /zedm/zed_node/pose_with_covariance
[ INFO] [1594656028.312992634]: Advertised on topic /zedm/zed_node/odom
[ INFO] [1594656028.313263890]: Advertised on topic /zedm/zed_node/path_odom
[ INFO] [1594656028.313532551]: Advertised on topic /zedm/zed_node/path_map
[ INFO] [1594656028.313829407]: Advertised on topic /zedm/zed_node/imu/data @ 400 Hz
[ INFO] [1594656028.314105709]: Advertised on topic /zedm/zed_node/imu/data_raw @ 400 Hz
[ INFO] [1594656028.314375379]: Advertised on topic /zedm/zed_node/imu/mag @ 50 Hz
[ INFO] [1594656028.314665315]: Advertised on topic /zedm/zed_node/left_cam_imu_transform [LATCHED]
[ INFO] [1594656028.317627663]: Static transform Sensor to Base [zedm_left_camera_frame -> base_link]
[ INFO] [1594656028.317658785]:  * Translation: {0.000,-0.030,0.000}
[ INFO] [1594656028.317682240]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1594656028.317711947]: Static transform Sensor to Camera Center [zedm_left_camera_frame -> zedm_camera_center]
[ INFO] [1594656028.317735240]:  * Translation: {0.000,-0.030,0.000}
[ INFO] [1594656028.317757193]:  * Rotation: {0.000,-0.000,0.000}
[ INFO] [1594656028.317785655]: Static transform Camera Center to Base [zedm_camera_center -> base_link]
[ INFO] [1594656028.317812516]:  * Translation: {0.000,0.000,0.000}
[ INFO] [1594656028.317834298]:  * Rotation: {0.000,-0.000,0.000}
^C[zedm/zed_node-3] killing on exit
[zedm/zedm_state_publisher-2] killing on exit
in Camera.cpp : void sl::Camera::close() at 84 : cuda error [10]: invalid device ordinal.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Myzhar, 2020-07-26:
Is VINS node publishing `odom` topic?
In that case you must disable the publishing of `odom` topic by the ZED node setting `publish_tf` to false

peter2527, 2020-07-26:
> Is VINS node publishing `odom` topic?
> In that case you must disable the publishing of `odom` topic by the ZED node setting `publish_tf` to false

yes! vins node publishing odom topic,so where should i set the ZED node setting publish_tf to flase?

Myzhar, 2020-07-26:
This is the correct place:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L47
but you can use also the example launch file provided for this scope:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/launch/zed_no_tf.launch
if you use it please remember to change the camera model to `zedm`

peter2527, 2020-07-26:
thank you very much,appreciate it!i will have a try.

peter2527, 2020-07-26:
> This is the correct place:
> https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L47
> but you can use also the example launch file provided for this scope: [...]

what a pity that the two ways you afford didn't work!

Myzhar, 2020-07-26:
Please post the output of the command `$ roswtf` while all the nodes are running

peter2527, 2020-07-26:
> Please post the output of the command `$ roswtf` while all the nodes are running

No errors or warnings
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:

/vins_estimator:
/vins_cam_switch
/feature_tracker/feature
/vins_restart
/vins_imu_switch
/zedm/zedm_state_publisher:
/zedm/joint_states
/rvizvisualisation:
/loop_fusion/pose_graph
/globalEstimator/car_model
/globalEstimator/global_path
/loop_fusion/match_image
/loop_fusion/pose_graph_path
/loop_fusion/camera_pose_visual
WARNING Received out-of-date/future transforms:

receiving transform from [/vins_estimator] that differed from ROS time by -7.666629553s

Myzhar, 2020-07-26:
Good, TFs are ok. You must work on VINS node to solve these warnings.
I suggest you to search on ROS answers or VINS repo

peter2527, 2020-07-26:
> Good, TFs are ok. You must work on VINS node to solve these warnings.
> I suggest you to search on ROS answers or VINS repo

OK

Myzhar, 2020-07-26:
One last thing. If no one node is publishing the `map` TF you should change the parameter `map_frame` in common.yaml to `odom` otherwise the root node of the TF tree is not existing

Myzhar, 2020-08-24:
@peter2527 can you please update the ticket or close it if you found a solution?

closed 2020-08-31 by Myzhar