stereolabs/zed-ros-wrapper #593: [BUG] dynamic parameter pub_frame_rate missing?

RGring, 2020-07-17:
**Describe the bug**
I can't set the dynamic parameter pub_frame_rate, because it is not listed.

A clear and concise description of what the bug is.

**Device information:**
 - OS: Ubuntu
 - OS 18.04
 - CPU: Jetson Xavier NX
 - ZED SDK Version: v4.3.2.0
 - ROS Wrapper version: master

**To Reproduce**
1. roslaunch zed_wrapper zed2.launch"/>_

2 rosrun dynamic_reconfigure dynparam list
```
/zed2/zed_node
/zed2/zed_node/depth/depth_registered/compressed
/zed2/zed_node/depth/depth_registered/compressedDepth
/zed2/zed_node/depth/depth_registered/theora
/zed2/zed_node/left/image_rect_color/compressed
/zed2/zed_node/left/image_rect_color/compressedDepth
/zed2/zed_node/left/image_rect_color/theora
/zed2/zed_node/left/image_rect_gray/compressed
/zed2/zed_node/left/image_rect_gray/compressedDepth
/zed2/zed_node/left/image_rect_gray/theora
/zed2/zed_node/left_raw/image_raw_color/compressed
/zed2/zed_node/left_raw/image_raw_color/compressedDepth
/zed2/zed_node/left_raw/image_raw_color/theora
/zed2/zed_node/left_raw/image_raw_gray/compressed
/zed2/zed_node/left_raw/image_raw_gray/compressedDepth
/zed2/zed_node/left_raw/image_raw_gray/theora
/zed2/zed_node/rgb/image_rect_color/compressed
/zed2/zed_node/rgb/image_rect_color/compressedDepth
/zed2/zed_node/rgb/image_rect_color/theora
/zed2/zed_node/rgb/image_rect_gray/compressed
/zed2/zed_node/rgb/image_rect_gray/compressedDepth
/zed2/zed_node/rgb/image_rect_gray/theora
/zed2/zed_node/rgb_raw/image_raw_color/compressed
/zed2/zed_node/rgb_raw/image_raw_color/compressedDepth
/zed2/zed_node/rgb_raw/image_raw_color/theora
/zed2/zed_node/rgb_raw/image_raw_gray/compressed
/zed2/zed_node/rgb_raw/image_raw_gray/compressedDepth
/zed2/zed_node/rgb_raw/image_raw_gray/theora
/zed2/zed_node/right/image_rect_color/compressed
/zed2/zed_node/right/image_rect_color/compressedDepth
/zed2/zed_node/right/image_rect_color/theora
/zed2/zed_node/right/image_rect_gray/compressed
/zed2/zed_node/right/image_rect_gray/compressedDepth
/zed2/zed_node/right/image_rect_gray/theora
/zed2/zed_node/right_raw/image_raw_color/compressed
/zed2/zed_node/right_raw/image_raw_color/compressedDepth
/zed2/zed_node/right_raw/image_raw_color/theora
/zed2/zed_node/right_raw/image_raw_gray/compressed
/zed2/zed_node/right_raw/image_raw_gray/compressedDepth
/zed2/zed_node/right_raw/image_raw_gray/theora
/zed2/zed_node/stereo/image_rect_color/compressed
/zed2/zed_node/stereo/image_rect_color/compressedDepth
/zed2/zed_node/stereo/image_rect_color/theora
/zed2/zed_node/stereo_raw/image_raw_color/compressed
/zed2/zed_node/stereo_raw/image_raw_color/compressedDepth
/zed2/zed_node/stereo_raw/image_raw_color/theora
```

Myzhar, 2020-07-17:
Hi @RGring 
I just checked using "rqt" and it is visible and it works as expected
[image]
It is available for the "/zed2/zed_node" node, that is the first of your list.

Just a remark: the command `rosrun dynamic_reconfigure dynparam list` does not list all the dynamic parameters, but it lists all the dynamically reconfigurable nodes.

RGring, 2020-07-20:
Alright, you are right. Thanks for the help.

closed 2020-07-20 by RGring