stereolabs/zed-ros-wrapper #594: [Question] What are final camera parameters after rectification?

JeyP4, 2020-07-21:
I understood that distortion correction is performed.

I am interested in what are the cx, cy, fx, fy after rectification `(/zed/zed_node/rgb/image_rect_color)` ? (Let's say for VGA)

Myzhar, 2020-07-21:
Hi @JeyP4 
you can find many good information searching for "Pin hole camera model".
- (cx, cy) is a principal point that is usually at the image center
- fx, fy are the focal lengths expressed in pixel units.

JeyP4, 2020-07-21:
> Hi @JeyP4
> you can find many good information searching for "Pin hole camera model".
> 
> * (cx, cy) is a principal point that is usually at the image center
> * fx, fy are the focal lengths expressed in pixel units.

Thanks
I am sorry, I was not clear enough in my question.

My question is in the config file I see these values. Which are required to perform undistort.
After undistort, distortion coefficients will vanish. But rectified image will have its new fx, fy, cx , cy.
May this wrappper publish somewhere new values?

Myzhar, 2020-07-21:
Each image topic has its own camera parameter associated.
e.g. 
`/zed2/zed_node/rgb/camera_info` and `/zed2/zed_node/rgb/image_rect_color` -> undistorted, but with fx, fy, cx, cy that you can find in the `K` parameter in the `camera_info` topic:
e.g. `K: [531.2548217773438, 0.0, 666.2540283203125, 0.0, 531.2548217773438, 349.59100341796875, 0.0, 0.0, 1.0]`
`fx = 531.2548217773438`
`fy = 531.2548217773438`
`cx = 666.2540283203125`
`cy = 349.59100341796875`

one example of complete `camera_info` topic:
```
---
header: 
  seq: 1
  stamp: 
    secs: 1595352953
    nsecs: 379747070
  frame_id: "zed2_left_camera_optical_frame"
height: 720
width: 1280
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [531.2548217773438, 0.0, 666.2540283203125, 0.0, 531.2548217773438, 349.59100341796875, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [531.2548217773438, 0.0, 666.2540283203125, 0.0, 0.0, 531.2548217773438, 349.59100341796875, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

JeyP4, 2020-07-21:
Thanks

Do you mean after undistort new `camera_matrix` = old `camera_matrix` ?

Myzhar, 2020-07-21:
No, I have not said that. Indeed you can compare the camera matrix before and after calibration and you will notice that they are different.
e.g. `/zed2/zed_node/rgb_raw/camera_info` (raw image) vs `/zed2/zed_node/rgb/camera_info` (undistorted image)

JeyP4, 2020-07-21:
Ok
Now it's clear.
I will check tomorrow.

 Thanks.

closed 2020-07-21 by JeyP4