stereolabs/zed-ros-wrapper #595: How can I customize my own ros-zed package to achieve the image capture function?

longningbo, 2020-07-22:
Hi, I am an beginner in ros, and I want to integrate the zed2 into my robot system. However, I found the ZED ROS wrapper include all the nodes and funcitons in the file folder, some nodes or functions are not needed in my system. And the file structure is very complex for me, so ,how can I customize my own ros package to ahcieve the left/right/depth image capture and IMU data acquire funciton in an alone ros package ?
Thanks very much.

Myzhar, 2020-07-22:
Hi @longningbo 
the only way of doing what you are planning to do is rewriting your own basilar ZED node implementing only the required features.
However the current ROS wrapper is highly modular and one feature is not enabled unless you are using it subscribing the releative topic or explicitly enabling it by changing one parameter.

longningbo, 2020-07-22:
ok, Thank you very much! 
However, it is a bad news.
@Myzhar

closed 2020-07-28 by longningbo