stereolabs/zed-ros-wrapper #596: [Question] Info about the actual FOV

RGring, 2020-07-22:
I figured, that the actual FOV decreases with increasing resolution. Can I retrieve somewhere the FOV of the configured camera in ros?

Myzhar, 2020-07-22:
Hi @RGring 
the information is available in the CameraParameter structure:
https://www.stereolabs.com/docs/api/structsl_1_1CameraParameters.html
but it's not provided in ROS.
We will add it with a future release of the wrapper. 

Stay tuned

Myzhar, 2020-07-22:
@RGring 
as they pointed out to me on [ROS answer](https://answers.ros.org/question/357799/is-there-a-topic-for-camera-fov-info/), adding FOV information is reduntant since you have camera focals `f_x` and `f_y` and frame sizes:
`fov_x = 2 * arctan( w / (2*f_x) )`
`fov_y = 2 * arctan( h / (2*f_y) )`

closed 2020-07-31 by RGring

Aayli, 2022-12-15:
Is it already available to check CameraParameters using ROS?

Myzhar, 2022-12-15:
Camera parameters are always available in the `camera_info` topic