stereolabs/zed-ros-wrapper #597: [Question] left/right/imu parameters available?

FaboNo, 2020-07-24:
Hi there, 
We would like to test the ZED2 with Kimera (https://github.com/MIT-SPARK/Kimera-VIO) and we need to provide the parameters for the IMU as well as the left and right cameras. Do you know where can I find them in the repo?
Thank you

Myzhar, 2020-07-24:
Hi @FaboNo 
I do not know the Kimera-VIO repository. What parameters does it need?
You can get the camera parameters for the current resolution in the `camera_info` topic or reading the `conf` file in the `/usr/local/zed/settings/` folder.

FaboNo, 2020-07-24:
@Myzhar sorry, I should give you some examples, I guess the format is the one provided by Kalibr, here are two examples of yaml files:

```
%YAML:1.0
# General sensor definitions.
camera_id: left_cam

# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [0.9999696204143095, 0.0026862211395019798, 0.007317271653524808, -0.035444853724645585,
         -0.0023778583642409027, 0.9991219422040934, -0.04182930068647981, 0.027123624325323524,
         -0.007423209417859271, 0.041810630494050476, 0.9990979767468398, -0.08347299538526054,
         0.0, 0.0, 0.0, 1.0]

# Camera specific definitions.
rate_hz: 30
resolution: [640, 480]
camera_model: pinhole
intrinsics: [385.18681587014777, 385.08894038512136, 323.08215719710927, 241.75845145344397] #fu, fv, cu, cv
distortion_model: equidistant
distortion_coefficients: [0.34058661190133155, 0.10891845722051097, 0.1099073078802829, -0.029097445495854555]
```

And for the IMU:
```
%YAML:1.0
# Type of IMU preintegration:
# 0: CombinedImuFactor
# 1: ImuFactor
imu_preintegration_type: 1

# Collected by Andrew Violette on 02/19/20
# Sensor extrinsics wrt. the body-frame.
T_BS:
  cols: 4
  rows: 4
  data: [1.0, 0.0, 0.0, 0.0,
         0.0, 1.0, 0.0, 0.0,
         0.0, 0.0, 1.0, 0.0,
         0.0, 0.0, 0.0, 1.0]
rate_hz: 300

# inertial sensor noise model parameters (static)
gyroscope_noise_density: 0.000488     # [ rad / s / sqrt(Hz) ]   ( gyro "white noise" )
gyroscope_random_walk: 4.88e-05       # [ rad / s^2 / sqrt(Hz) ] ( gyro bias diffusion )
accelerometer_noise_density: 0.00147  # [ m / s^2 / sqrt(Hz) ]   ( accel "white noise" )
accelerometer_random_walk: 0.00147    # [ m / s^3 / sqrt(Hz) ].  ( accel bias diffusion )

# Extra parameters
imu_integration_sigma: 1.0e-8
imu_time_shift: 0.0
n_gravity: [0.0, 0.0, -9.81]

```

Myzhar, 2020-07-24:
The `# inertial sensor noise model parameters` are available in the [`SensorParameters` data structure](https://www.stereolabs.com/docs/api/structsl_1_1SensorParameters.html).
They are not exposed in the ROS node, but you can get them simply modifying the [`Sensor Data` tutorial](https://github.com/stereolabs/zed-examples/tree/master/tutorials/tutorial%207%20-%20sensor%20data)

The `# Sensor extrinsics wrt. the body-frame.` is available in TF, but you can get the values reading the console log when the node starts:
for example:
```
[ INFO] [1595584347.430331450]: Camera-IMU Transform: 
 5596CB1354C0
0.999626 0.025546 0.009792 -0.002000
-0.025588 0.999664 0.004221 -0.023000
-0.009681 -0.004470 0.999943 -0.002000
0.000000 0.000000 0.000000 1.00000```

The only missing information is a precise `imu_time_shift`, you need to run Kalibr for that.

closed 2020-07-24 by FaboNo