stereolabs/zed-ros-wrapper #598: [Question]  Rotating X Y axis around Z

jaronski, 2020-07-24:
Hello everyone,
just got my ZED2 camera and i would like to get the distance of objects in Y direction the Width in X and Z can stay as it is? Can anyone point me in the right direction where to change the coordinate system ? i rotated the cam_yaw parameter in the zed2.launch bei 90 Degrees and rviz is showing the orientation i expect. When i ceck the values of the 3D Boundingbox unfortunatly  the distance to the detected person is in X and not in Y. Can anyone please help me with this?

best regards and thanks in advance

Myzhar, 2020-07-24:
Hi @jaronski 
in the ZED ROS wrapper we use the `sl::COORDINATE_SYSTEM::RIGHT_HANDED_Z_UP_X_FWD` coordinate frame setting because it reflects the ROS conventions as described in the [REP103](https://www.ros.org/reps/rep-0103.html#coordinate-frame-conventions).

You can change it modifying the following row in the code:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L219
using instead the cordinate frame that you require (see https://www.stereolabs.com/docs/api/group__Core__group.html#ga1114207cbac18c1c0e4f77a6b36a8cb2)
but be aware that this can generate strange unexpected behaviours.

jaronski, 2020-07-24:
thanks for the info

Myzhar, 2020-08-24:
@jaronski can you please update the ticket or close it if you found a solution?

closed 2020-08-24 by jaronski