stereolabs/zed-ros-wrapper #599: imu:Status Error,Transform[sender=unknown_publisher]

peter2527, 2020-07-26:
when i launch"roslaunch zed_display_rviz display_zedm.launch"  in the terminal
errors comes with imu:
Transform [sender=unknown_publisher]
For frame [zedm_imu_link]: No transform to fixed frame [map]. TF error: [Lookup would require extrapolation into the future. Requested time 1595746740.922944771 but the latest data is at time 1595746740.879109718, when looking up transform from frame [zedm_imu_link] to frame [map]]

ubuntu 18.04 
cuda10.0 zedsdk 3.2.1

Myzhar, 2020-07-26:
Hi @peter2527 
If the error is inside Rviz you can ignore it. The IMU frame is published at 400Hz, while all the other topics are published at max 100Hz, so Rviz "goes crazy".
You can avoid the error enabling image imu synchronization

peter2527, 2020-07-26:
> Hi @peter2527
> If the error is inside Rviz you can ignore it. The IMU frame is published at 400Hz, while all the other topics are published at max 100Hz, so Rviz "goes crazy".
> You can avoid the error enabling image imu synchronization

thanks!
but it seems affect other application,please check out my issue"https://github.com/stereolabs/zed-ros-wrapper/issues/589"
otherwise the sdk showed  the version 3.2.0 repaire the bug "Fixed bug in IMU orientation covariance, leading to NaN occurences."
it confused me,may i beg your help?

Myzhar, 2020-07-26:
It seems that you have problem of TF overlapping. In these case the command `$ roswtf` while all the nodes are running can give an idea of what is the wrong configuration

peter2527, 2020-07-26:
> It seems that you have problem of TF overlapping. In these case the command `$ roswtf` while all the nodes are running can give an idea of what is the wrong configuration

Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 2 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /vins_estimator:
   * /vins_cam_switch
   * /feature_tracker/feature
   * /vins_restart
   * /vins_imu_switch
 * /zedm/zedm_state_publisher:
   * /zedm/joint_states
 * /rvizvisualisation:
   * /loop_fusion/pose_graph
   * /globalEstimator/car_model
   * /globalEstimator/global_path
   * /loop_fusion/match_image
   * /loop_fusion/pose_graph_path
   * /loop_fusion/camera_pose_visual

WARNING Received out-of-date/future transforms:
 * receiving transform from [/vins_estimator] that differed from ROS time by -7.666629553s

Myzhar, 2020-08-24:
@peter2527 can you please update the ticket or close it if you found a solution?

closed 2020-08-31 by Myzhar