stereolabs/zed-ros-wrapper #600: rtabmap won't work on Jetson with zed-ros wrapper

AndreV84, 2020-07-27:
https://github.com/stereolabs/zed-docker/issues/24#issue-665892129
https://github.com/stereolabs/zed-docker/issues/22#issuecomment-664062182

_Originally posted by @AndreV84 in https://github.com/stereolabs/zed-ros-wrapper/issues/559#issuecomment-664432361_

AndreV84, 2020-07-27:
with the updated Dockerfile that uses 3.2.,1 sdk it seems to run but still warnings;
Shall the launchfile be adjusated for use with zed2 camera? with zed mini?
[Dockerfile.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4982972/Dockerfile.txt)

AndreV84, 2020-07-27:
running with
docker run -it --rm --net=host   -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE zed:latest /bin/bash -c '.  /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch zed_rtabmap_example zed_rtabmap.launch'
The output is:
```
 docker run -it --rm --net=host   -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE zed:latest /bin/bash -c '.  /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch zed_rtabmap_example zed_rtabmap.launch'
... logging to /root/.ros/log/0be83062-d021-11ea-9268-d6aad4984c45/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:40573/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6
 * /zed/rtabmap/Grid/3D: True
 * /zed/rtabmap/Grid/CellSize: 0.05
 * /zed/rtabmap/Grid/ClusterRadius: 0.1
 * /zed/rtabmap/Grid/DepthDecimation: 1
 * /zed/rtabmap/Grid/DepthRoiRatios: [0.0, 0.0, 0.0, 0.0]
 * /zed/rtabmap/Grid/FlatObstacleDetected: True
 * /zed/rtabmap/Grid/FootprintHeight: 2.0
 * /zed/rtabmap/Grid/FootprintLength: 0.18
 * /zed/rtabmap/Grid/FootprintWidth: 0.18
 * /zed/rtabmap/Grid/FromDepth: True
 * /zed/rtabmap/Grid/GroundIsObstacle: False
 * /zed/rtabmap/Grid/MapFrameProjection: False
 * /zed/rtabmap/Grid/MaxGroundAngle: 30.0
 * /zed/rtabmap/Grid/MaxGroundHeight: -0.4
 * /zed/rtabmap/Grid/MaxObstacleHeight: 0.1
 * /zed/rtabmap/Grid/MinClusterSize: 3
 * /zed/rtabmap/Grid/MinGroundHeight: -0.5
 * /zed/rtabmap/Grid/NoiseFilteringMinNeighbors: 5
 * /zed/rtabmap/Grid/NoiseFilteringRadius: 0.1
 * /zed/rtabmap/Grid/NormalK: 20
 * /zed/rtabmap/Grid/NormalsSegmentation: False
 * /zed/rtabmap/Grid/PreVoxelFiltering: True
 * /zed/rtabmap/Grid/RangeMax: 3.0
 * /zed/rtabmap/Grid/RangeMin: 0.7
 * /zed/rtabmap/Grid/RayTracing: True
 * /zed/rtabmap/GridGlobal/Eroded: False
 * /zed/rtabmap/GridGlobal/FootprintRadius: 0.18
 * /zed/rtabmap/GridGlobal/FullUpdate: True
 * /zed/rtabmap/GridGlobal/MaxNodes: 0
 * /zed/rtabmap/GridGlobal/MinSize: 1.0
 * /zed/rtabmap/GridGlobal/OccupancyThr: 0.55
 * /zed/rtabmap/GridGlobal/ProbClampingMax: 0.971
 * /zed/rtabmap/GridGlobal/ProbClampingMin: 0.1192
 * /zed/rtabmap/GridGlobal/ProbHit: 0.7
 * /zed/rtabmap/GridGlobal/ProbMiss: 0.4
 * /zed/rtabmap/GridGlobal/UpdateError: 0.01
 * /zed/rtabmap/Mem/IncrementalMemory: true
 * /zed/rtabmap/Mem/InitWMWithAllNodes: false
 * /zed/rtabmap/approx_sync: True
 * /zed/rtabmap/config_path: ~/.ros/rtabmap.cfg
 * /zed/rtabmap/database_path: ~/.ros/rtabmap.db
 * /zed/rtabmap/frame_id: base_link
 * /zed/rtabmap/map_frame_id: map
 * /zed/rtabmap/odom_frame_id: odom
 * /zed/rtabmap/odom_sensor_sync: False
 * /zed/rtabmap/odom_tf_angular_variance: 0.001
 * /zed/rtabmap/odom_tf_linear_variance: 0.001
 * /zed/rtabmap/publish_tf: False
 * /zed/rtabmap/queue_size: 10
 * /zed/rtabmap/scan_normal_k: 0
 * /zed/rtabmap/subscribe_depth: True
 * /zed/rtabmap/subscribe_odom_info: False
[... 215 lines cut ...]
   /zed/zed_node/rgb/image_rect_color,
   /zed/zed_node/depth/depth_registered,
   /zed/zed_node/rgb/camera_info
[ WARN] [1595865141.261341593]: /zed/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/zed/rtabmap subscribed to (approx sync):
   /zed/zed_node/rgb/image_rect_color,
   /zed/zed_node/depth/depth_registered,
   /zed/zed_node/rgb/camera_info
[ WARN] [1595865146.261966900]: /zed/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/zed/rtabmap subscribed to (approx sync):
   /zed/zed_node/rgb/image_rect_color,
   /zed/zed_node/depth/depth_registered,
   /zed/zed_node/rgb/camera_info
[ WARN] [1595865151.262869271]: /zed/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/zed/rtabmap subscribed to (approx sync):
   /zed/zed_node/rgb/image_rect_color,
   /zed/zed_node/depth/depth_registered,
   /zed/zed_node/rgb/camera_info
[ WARN] [1595865156.263264166]: /zed/rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/zed/rtabmap subscribed to (approx sync):
   /zed/zed_node/rgb/image_rect_color,
   /zed/zed_node/depth/depth_registered,
   /zed/zed_node/rgb/camera_info

```

Myzhar, 2020-08-24:
@AndreV84 can you please update the ticket or close it if you found a solution?

closed 2020-08-31 by Myzhar

AndreV84, 2020-08-31:
@Myzhar Thank you for following up.
In my case complication was that database viewer won't draw 3dmap.