stereolabs/zed-ros-wrapper #601: zed ros wrapper doesn't start on Jetson AGX using the dockerfile below

AndreV84, 2020-07-27:
```
```
docker run -it --rm --net=host   -e DISPLAY=$DISPLAY --ipc=host --privileged -v /tmp/.X11-unix/:/tmp/.X11-unix/ -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE zed:latest /bin/bash -c '.  /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch zed_wrapper zed2.launch'
... logging to /root/.ros/log/5194cbac-d026-11ea-bc1d-00044bcb9a37/roslaunch-jetson-agx001-54.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jetson-agx001:40645/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: True
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 15
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path: 
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed2/zed_node/pos_tracking/floor_alignment: False
 * /zed2/zed_node/pos_tracking/imu_fusion: True
 * /zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed2/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed2/zed_node/pos_tracking/map_frame: map
 * /zed2/zed_node/pos_tracking/odometry_frame: odom
 * /zed2/zed_node/pos_tracking/path_max_count: -1
 * /zed2/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed2/zed_node/pos_tracking/publish_map_tf: True
 * /zed2/zed_node/pos_tracking/publish_tf: True
 * /zed2/zed_node/pos_tracking/two_d_mode: False
 * /zed2/zed_node/pub_frame_rate: 15.0
 * /zed2/zed_node/saturation: 4
 * /zed2/zed_node/sensors/sensors_timestamp_sync: False
 * /zed2/zed_node/sharpness: 4
 * /zed2/zed_node/stream: 
 * /zed2/zed_node/svo_file: 
 * /zed2/zed_node/video/extrinsic_in_camera_frame: True
 * /zed2/zed_node/video/img_downsample_factor: 1.0
 * /zed2/zed_node/whitebalance_temperature: 42

NODES
  /zed2/
    zed2_state_publisher (robot_state_publisher/robot_state_publisher)
    zed_node (zed_wrapper/zed_wrapper_node)

auto-starting new master
process[master]: started with pid [67]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5194cbac-d026-11ea-bc1d-00044bcb9a37
process[rosout-1]: started with pid [78]
started core service [/rosout]
process[zed2/zed2_state_publisher-2]: started with pid [85]
process[zed2/zed_node-3]: started with pid [86]
[ INFO] [1595867362.174297071]: Initializing nodelet with 8 worker threads.
[ERROR] [1595867362.207544225]: Failed to load nodelet [/zed2/zed_node] of type [zed_nodelets/ZEDWrapperNodelet] even after refreshing the cache: Failed to load library /home/nvidia/ros_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)
[ERROR] [1595867362.207671750]: The error before refreshing the cache was: Failed to load library /home/nvidia/ros_ws/devel/lib//libZEDNodelets.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libcuda.so.1: cannot open shared object file: No such file or directory)
^C[zed2/zed_node-3] killing on exit
[zed2/zed2_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
[dockerfile.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4983189/dockerfile.txt)

AndreV84, 2020-07-27:
[dockerfile.txt](https://github.com/stereolabs/zed-ros-wrapper/files/4983191/dockerfile.txt)

AndreV84, 2020-07-27:
running with different arguments will cause driver cuda issue:
```
docker run -it --rm --runtime nvidia --privileged -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix  -v /tmp/argus_socket:/tmp/argus_socket --cap-add SYS_PTRACE zed:latest /bin/bash -c '. /opt/ros/melodic/setup.bash; source devel/setup.bash; roslaunch zed_wrapper zed2.launch'

... logging to /root/.ros/log/14a42da4-d027-11ea-8ef8-0242ac110002/roslaunch-0cfd3144e784-55.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://0cfd3144e784:40209/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.6
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: True
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 15
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path: 
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed2/zed_node/pos_tracking/floor_alignment: False
 * /zed2/zed_node/pos_tracking/imu_fusion: True
 * /zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed2/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed2/zed_node/pos_tracking/map_frame: map
 * /zed2/zed_node/pos_tracking/odometry_frame: odom
 * /zed2/zed_node/pos_tracking/path_max_count: -1
 * /zed2/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed2/zed_node/pos_tracking/publish_map_tf: True
 * /zed2/zed_node/pos_tracking/publish_tf: True
[... 104 lines cut ...]
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 732 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::SkeletonDetectorHandler::clear() at 831 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in Camera.cpp : void sl::Camera::close() at 181 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
[ INFO] [1595867691.569035546]: ZED connection -> NVIDIA DRIVER OUT OF DATE
^C[zed2/zed_node-3] killing on exit
[zed2/zed2_state_publisher-2] killing on exit
in Camera.cpp : void sl::Camera::close() at 84 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
in Camera.cpp : void sl::Camera::close() at 134 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 679 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 703 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 708 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 714 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 717 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 723 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::ObjectsDetectorHandler::clear() at 732 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in CameraUtils.hpp : void sl::SkeletonDetectorHandler::clear() at 831 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in Camera.cpp : void sl::Camera::close() at 181 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
in Camera.cpp : void sl::Camera::close() at 84 : cuda error [35]: CUDA driver version is insufficient for CUDA runtime version.
[rosout-1] killing on exit
^X[master] killing on exit
^Zshutting down processing monitor...
... shutting down processing monitor complete
done
```

AndreV84, 2020-07-27:
My concerns are

- run on Jetpack4.4 with zed sdk 3.2.1 within docker, the rtabmap
- run the default ros wrapper node
could you help with both? any? neither of the two?

Myzhar, 2020-08-24:
@AndreV84 can you please update the ticket or close it if you found a solution?

closed 2020-08-31 by Myzhar

AndreV84, 2020-08-31:
Thank you for following up!
I am just wondering if you are able to run the 3d cloud from database viewer from the docker.